DomiStyle/esphome-dlms-meter #1: Support for other smart meter in austria

greenMikeEU, 2021-07-20:
Hello my name is Michael i have an samrt Meter dein Evn (Netz Niederösterreich). 

I think this Meter works same as yours. Can you Help me to set Up? I had some questions. 

You can write me an e-mail.
mreitbauer@gmx.net

BR Michael

Warp9999, 2021-07-22:
It should work for all KSMWest (Kooperation Smartmeter West) Providers.
(Vorarlberg, Tirol, Salzburg)

I am in Vorarlberg and i can confirm it is working with "Vorarlberger Energienetze" Smartmeters.
https://www.computerbase.de/forum/threads/smartmeter-kundenschnittstelle-aus-der-hoelle.2031259/#post-25810438

Thank you Domistyle for your Work.

But i am not completely happy to run it over an ESP32.
Would it be possible to change te code to run natively on Linux with an USB MBus Slave Adadpter?

DomiStyle, 2021-07-22:
@greenMikeEU Can you post the exact make and model of your smart meter since yours is not from Kaifa?

@Warp9999 Thanks, I added the other providers to the supported list.

> Would it be possible to change te code to run natively on Linux with an USB MBus Slave Adadpter?

It's a C++ library so it can run everywhere where C++ can be compiled. However, some work is needed to remove the calls to the ESPHome API.
Also serial data needs to be fed to the packet decoder. Currently ESPHome takes care of that.

The handling of the packet decoding/decrypting starts here: https://github.com/DomiStyle/esphome-dlms-meter/blob/master/espdm.cpp#L34

Warp9999, 2021-07-22:
Sorry, i cannot code something. I have no Clue about this.

I only have some Skills with Siemens PLC and Node-Red.

There are  also some Smatmeter Plugins for Node-Red, but not for mine.
A Node Red Plugin would be great.

DomiStyle, 2021-07-22:
It's possible to adapt the library for Linux, but somebody else will have to figure it out.

andyboeh, 2021-08-17:
I just found your project while researching a problem with my implementation for the AMIS smart meter in Upper Austria. My problem is that I sometimes get very high readings, like 300.000.000,00 W instantaneous power while the packet checksum is correct and the timestamp decodes properly. Are you also getting such spurious readings?

While my meter is a Siemens TD-3511, the used protocol seems very similar: Encrypted M-Bus over IR. You can find my code at https://github.com/andyboeh/esphome-amis

DomiStyle, 2021-08-17:
@andyboeh No, the highest value I have received in the last 3 months was 12kW, which is plausible.

andyboeh, 2021-08-17:
@DomiStyle OK, thanks. I might have found it: I did not parse the frame properly, but used static indices for the data fields. Sometimes, the meter sends other data (or the data is corrupt) and, upon proper parsing, this data is excluded.

TomBoehm, 2021-08-31:
In Lower Austria, EVN supplied me a Kaifa MA309M, but they sealed the M-Bus connection. How is the situation at your site? Did you just break the seal?

DomiStyle, 2021-08-31:
@TomBoehm Even though it is misleading the actual M-Bus connector is located above the sealed connector that says M-Bus.

It's not necessary to break any seals.

[image]

TomBoehm, 2021-09-01:
@DomiStyle thank you!

micronano0, 2021-09-02:
@DomiStyle 
Hallo Dominik, ich habe Teile deines Codes (Obis Teil) in meiner ESP8266 Umsetzung verwendet. Ist es gestattet meine Umsetzung (mit einem Teil Code von dir) auf Github zu veröffentlichen?
In meiner Umsetzung mache ich einen Hotspot auf dem Handy und verbinde mich mit dem ESP8266. Dann kann ich auf der Webseite den Key setzen und die Daten vom Zähler lesen.

PS: ich finde deinen Code genial programmiert!

DomiStyle, 2021-09-02:
@micronano0 Klar, der Code ist unter MIT lizenziert, du kannst damit alles machen was du willst solange du die Lizenz und Urheberrechtsbeschreibung intakt lässt.

Siehe hier:
https://github.com/DomiStyle/esphome-dlms-meter/blob/master/LICENSE

micronano0, 2021-09-02:
@DomiStyle 
alles klar, vielen Dank!

sheminasalam, 2022-01-14:
Will this work with Landis+Gyr E250 Net meter? It is DLMS complaint. In theory all DLMS complaint meter should be using the same protocol and hardware...right? Also this is how my meter com port is labelled. Hope some one would be able to explain how I should do the wiring also.

[image]

madmaiike, 2023-01-26:
Hi there Dominik, 

The electricity company (Israel) changed my meter today for Kaifa MA309M  - and I'm in the process of going back to Home Assistant :) 

I have a P1 port on the right (rj11 connector it seems) 

https://ibb.co/Db6dykf

Now will I be able to read it using ESP Home ? How do I find the key ? 

Thanks 

Michael