DomiStyle/esphome-dlms-meter #2: Payload length is too big - VKW Vorarlberg

color86, 2021-09-11:
Hello,

i have an ESP32 with ESPHome running. 
I entered the correct key but I get the following debug messages.

[22:15:25][D][espdm:040]: Handling packet
[22:15:25][E][espdm:062]: Payload length 1 is too big
[22:15:40][D][espdm:040]: Handling packet
[22:15:40][E][espdm:062]: Payload length 1 is too big
[22:15:55][D][espdm:040]: Handling packet
[22:15:55][E][espdm:062]: Payload length 1 is too big
[22:16:10][D][espdm:040]: Handling packet
[22:16:10][E][espdm:050]: Payload length is too big for received data
[22:16:25][E][espdm:036]: Received packet with invalid size
[22:16:25][E][espdm:036]: Received packet with invalid size
[22:16:40][D][espdm:040]: Handling packet
[22:16:40][E][espdm:062]: Payload length 1 is too big
[22:16:55][D][espdm:040]: Handling packet
[22:16:55][E][espdm:050]: Payload length is too big for received data
[22:17:10][D][espdm:040]: Handling packet
[22:17:10][E][espdm:062]: Payload length 1 is too big
[22:17:25][D][espdm:040]: Handling packet
[22:17:25][E][espdm:050]: Payload length is too big for received data
[22:17:40][D][espdm:040]: Handling packet

Do you have any idea whats wrong? Thanks

DomiStyle, 2021-09-11:
Can you send me the full packet + decryption key to the mail address on my profile?

color86, 2021-09-11:
i can't see a mail adress at your profile.

DomiStyle, 2021-09-11:
Strange, it's git@secnd.me .

color86, 2021-09-17:
i sent you an email.

this was the serial paket
strange that the serial paket starts with "CB"

Debug Paket
[D][espdm:455]:  CB 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 01 55 21 FC D9 ED AC E9 21 1C 7B 24 83 FF BC DB 93 8C 5C 93 EE EC 85 A4 52 B5 2B D2 21 F8 08 BC FD BB 31 3B A9 B7 F9 59 4A 2E DA B2 64 3D 31 B3 E5 EB 89 6E B5 F0 73 41 D3 E3 3E 08 23 82 A5 BA 4F 93 DA E9 2B 4A 90 EA AC 53 EC 5D 50 F4 07 36 8F C4 D7 A8 E5 E8 A8 86 C5 72 B6 D8 EC CE 0C 19 11 94 42 F7 C2 92 79 F9 32 F1 86 FF B1 03 04 16 DB 5F EC 27 4B D6 D9 71 BA 51 1D 40 23 02 BA A0 BB FD 47 61 2B CB EE BF A6 5E DF E3 75 70 03 E6 82 50 AE 7A 7A 0D 43 39 12 AB D4 3D D1 26 D0 F6 63 59 E4 1F 52 B8 08 BC FF 8B AD 7B 8E 34 A4 3F 15 FE 76 8B 44 7D 8F BD B8 98 99 81 F4 98 A4 60 8D 37 9C 2F 18 8A 4D C9 AF BA DC DB F7 5B 09 58 62 76 1B D8 AE 70 8A 59 16 68 72 72 68 53 FF 11 01 67 D5 73 BF AB D5 83 94 49 CE 58 7A 0E 0D 95 6B AA 99 48 DE 32 F7 2C 1D 2E DF 37 D7 73 85 91 42 17 7F 77 D6 56 30 1F D6 4D F9 4F 8E 17 C6 FA EF 92 79 26 5F 73 32 E2 53 5E 46 E5 F4 6F 0E E9 A6 10 00 AD 2D 84 04 62 F8 A4 53 18 4F FF 7B 59 F1 A2 59 87 C1 1B BF 13 AB 72 1D 8B 37 A6 36 52 F2 78 83 B2 9C 45 62 CF 98 8A 97 16 80 8B 06 7E 62 CD FC 04 81 10 B9 25 C8 9B F1 56 B6 16 39 49 C7 10 16 68 72 72 68 53 FF 11 01 67 4C 3B 43 58 D8 6E 77 75 AF 6B 29 22 1B 39 86 13 F9 2E 9A 12 DE 03 F4 AE 21 44 74 EC E5 08 45 CB 91 11 96 79 D4 59 FD D8 AF 48 C1 66 51 C4 FF AD 5E A5 FA 93 D3 A3 FD 40 17 4B CA 64 2D E5 A4 0E 3B 28 D0 0C FD 2B 88 CB 05 FF BF A4 25 97 D6 32 AB E3 E7 39 1E 59 E3 71 73 84 91 B3 49 EF 6C 8A 75 87 9E E1 A6 BF 71 16 64 E8 F3 16 AD 8D 5A FF FF 39 F1 AB 97 CD 3A 0C 97 9D BB 52 73 C8 16 34 41 3E 16 BE E9 73 98 4E 16 2C EF F9 74 33 16 73 5A AF 97 9D EB D2 CE CF 6B 01 81 43 0B 29 6A 85 AF 64 A1 B3 EE A5 CF E1 0B 56 54 32 65 2D 0E B8 47 D3 A7 58 E6 BE 51 85 12 FB C2 D7 20 D9 E7 88 9A 52 D5 FD 79 C8 99 FA 08 51 43 41 4A 54 0C 8F F6 FF CC FD 9A DD BD 63 78 D9 BA 1D 96 A5 FB 08 F7 72 3B D9 E4 82 DE 68 23 96 6B 8C F0 6D DD 13 DA F6 87 26 DE 59 73 EC 9A 07 18 EA 83 E8 86 0D 08 4A BB B5 4A 91 B2 20 B0 E1 72 81 36 A3 E7 5A 47 36 4F 4D 46 BE F2 4C F0 38 02 28 CA D5 1F 00 C

DomiStyle, 2021-09-19:
@color86 Something seems garbled here, your packet is too long and starts with the wrong byte.

Can you try logging over a longer period (like 10 or 20 packets) and posting the log here?

color86, 2021-09-19:
Hello,

here is the new log.
Thanks

[16:12:58][D][api.connection:735]: Home Assistant 2021.9.6 (192.168.1.139): Connected successfully
[16:13:00][D][espdm:040]: Handling packet
[16:13:00][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 C1 55 07 37 B4 6A 35 3A E1 37 83 91 03 0E BE F2 EF 8D E6 02 30 FF 80 F3 26 FA 0D 2A 69 2C 82 D6 D8 2C 99 1C DD C7 51 E7 82 E1 62 59 D0 B2 A8 1A C7 EB F5 55 6F 24 C1 7F 53 F9 68 95 17 FC 51 65 A0 FF CE 67 32 15 72 85 CB A4 CD DE 62 E3 F9 5B 23 4D 69 78 0B 9D ED 6E E1 8D C9 36 20 13 D2 27 AB 78 CF A5 2A 68 3E F0 F8 05 21 FE AB 48 D1 E9 47 FF F5 54 54 89 B3 28 8F 09 E6 DB 4E 24 FD 45 9A 58 8B 08 FA 16 3A F2 53 94 C7 92 7B F1 2B
[16:13:00][E][espdm:051]: Payload length is too big for received data
[16:13:15][D][espdm:040]: Handling packet
[16:13:15][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 5F B8 E3 FE C2 8A 76 3D 53 56 65 88 04 FA 2F CA 5D D9 A1 4D 70 3E CB 63 A5 E9 A8 61 ED 16 F2 08 C9 C8 A2 A8 3A B7 01 EC FC F3 37 2A 67 7D C9 2A 02 E6 E2 2D FA 1F FF FF 7F FF FF 93 55 55 2E FF BD 71 9C 5D 5C 45 79 38 BF 7B C3 6D C1 35 5F 3D 4C E4 06 E5 47 AF B1 2C AB 1F 58 9E 21 AD 31 91 98 F5 B6 BD CF 89 AD 65 EC F3 69 1D 35 3F 68 CE F0 36 35 11 A4 99 3A AE 0E 5B 73 24 5A 54 34 AD EE 44 CB 20 7B 8E D9 13 22 DA A2 83 7B 2A
[16:13:15][E][espdm:051]: Payload length is too big for received data
[16:13:30][D][espdm:040]: Handling packet
[16:13:30][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 C9 E7 D4 D0 E3 23 0E 10 D6 72 E5 09 17 56 6E 43 4B B7 39 28 E3 06 F3 FC FD DB AB 49 52 87 AB 9B 40 B0 0B EC C0 44 3F F5 C9 98 51 45 DA 10 CF A5 B8 D4 72 B7 2B 5D 71 3B D1 42 62 C3 D4 FE E6 DD 3F 0B C2 B6 F7 11 19 B3 91 BF 04 E1 EF 62 B9 22 44 8E 50 93 BF 3F 06 7C 1F 05 F0 DB BA 3A 50 66 6D AD 42 77 2D A6 7B C6 16 D1 9E 27 D1 22 CB 83 E2 F7 77 F6 2F B8 BF E5 F6 D3 5C 0D 4A 98 01 34 1C 49 3F B2 3F A3 E6 D3 ED 42 86 5D 4E 7A
[16:13:30][E][espdm:051]: Payload length is too big for received data
[16:13:45][D][espdm:040]: Handling packet
[16:13:45][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 F1 D1 35 B9 FC 0A C9 F0 E5 B2 02 3E FE D2 EE FF D3 5B 46 96 E6 12 F5 3A 5C B3 ED 1C DF 20 3E 44 79 B7 F8 3F FF 7E 22 E3 AE 8B 0D F2 6C E0 FA EC FF 32 3B 29 55 49 9A 6A 18 53 91 ED 78 E0 74 37 B6 11 98 CA 05 27 B6 C8 57 34 F5 81 9A 73 87 2D 08 07 76 8D 78 0F ED BF 6E 02 FA 62 D4 49 7D 33 FA 66 67 B3 5C FA 22 6B B9 46 74 B6 F4 3F 90 DD 71 6B CB EA 5D 29 19 52 CD D3 E8 E8 C5 C2 4E 92 85 B0 94 BB 6F 44 8F 1E 1F 8A F8 DD AD 02
[16:13:45][E][espdm:051]: Payload length is too big for received data
[16:14:00][D][espdm:040]: Handling packet
[16:14:00][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 EE D3 2D F5 9C 5D AE 16 EE 27 92 C9 AB F0 08 41 93 3E 39 D9 B2 0E 5A DD 8E 2B 95 41 6F DC 20 9C FF EB EA E7 28 78 4E D5 24 B7 67 14 7C C5 FE 5A CD 4F F2 FF 87 FB 32 CC 55 AE 6E FE 8F 3C 2E 3B B5 56 08 DF 4E 82 E5 E4 C7 8B 1E 69 17 C3 A3 B5 D4 EE C1 31 A2 4B C7 45 31 6C E2 F6 53 B2 D5 BC 0F 9B 90 DD FF 0A 0C BF 48 40 F8 FD 65 F6 E4 E6 88 52 C5 7C 99 01 46 F1 39 EA BF 6F B8 49 52 95 7E 61 8E 65 15 3E 37 17 94 D1 3C 41 6F 3E
[16:14:00][E][espdm:051]: Payload length is too big for received data
[16:14:15][D][espdm:040]: Handling packet
[16:14:15][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 FD FF FF E3 D3 E4 0F BF 38 6C 52 36 33 B1 76 53 A0 35 85 9B C3 5E 35 F0 13 19 46 DC 4A 4F 04 EE 07 CF 33 3E C5 5A DF 35 29 7C B7 BA 17 E4 7F 08 7E 61 85 85 A6 76 B6 32 D1 4F E3 F2 C2 C8 45 FF 17 BE 4C 02 87 AD 0F 9F 20 30 CF AC CC F3 85 EE 96 B7 F7 F8 7A F0 DD 9B CB 15 03 F4 48 DA 50 01 2A 37 AF 02 68 30 F3 FF FF 4D FE 9A 47 0B FF FC F5 47 FC 9E 37 93 E5 4B 56 37 63 2A 45 4B 0A 0B 5D E7 FC C1 B4 14 C7 2B BF 27 39 EC D2 75
[16:14:15][E][espdm:051]: Payload length is too big for received data
[16:14:30][D][espdm:040]: Handling packet
[16:14:30][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 5B 13 19 79 06 0F FB 11 7A DA 3D 4B DE BA BE E9 57 01 63 9A 09 B1 8F 7E 03 98 79 54 B9 7F 89 EB 7F 49 6E CF 4B 0E D3 5C CC 50 72 0A 11 B7 33 9F DF D9 05 FC 41 05 2F 2F DD 31 C5 11 FF 56 35 8A 24 4F 65 A1 68 3F 92 AB B8 EB 5C 2B 5F 82 24 EB CF CF 89 A4 D9 10 FD 98 C4 65 B1 69 23 36 0D 75 7F 74 23 16 43 66 DF 32 28 B8 0B F5 70 1D 38 2E 5F 38 84 FC CA 25 B5 15 B1 E2 F9 3D 73 47 CE 1F BA C5 46 DC 65 48 D7 B7 D6 FF ED 4D 9F 4A
[16:14:30][E][espdm:051]: Payload length is too big for received data
[16:14:45][D][espdm:040]: Handling packet
[16:14:45][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 F5 1F DE 1A A2 D7 84 CB 89 44 E5 25 AF 06 35 FD 73 2D 69 4A 63 F7 E4 53 F5 AC D2 AA D9 92 4E 0C 83 B3 3E 5F AA 4F 02 8F EC 44 8E 67 45 8F DF FA 1B D7 75 C7 40 1D 3E 5B E6 FA A9 D0 E8 5B 57 DF E8 B3 86 4D A4 29 4A A7 F0 B0 90 58 AE B4 89 D1 B5 2B 02 48 1F 96 2F 78 48 73 FC 62 C3 BF 9C 3B DD 4D 08 60 D2 E8 26 F6 E5 26 B2 6F 11 C1 7D 5C 5A 4E D9 55 A8 60 53 FD FD C2 F1 6C DA 11 12 6D FF F1 9D 6A 19 BE 06 59 FE 71 FF FF FC FB
[16:14:45][E][espdm:051]: Payload length is too big for received data
[16:15:00][D][espdm:040]: Handling packet
[16:15:00][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 FD FC FC A2 1B 49 A1 34 95 FA 57 7B 73 7E D2 B4 90 32 B7 B2 14 FE 79 FF FF DF AA 01 CC C3 EE 5C 8D 58 EC 76 E7 E7 63 81 99 1A AD D1 F4 C2 7C 38 3D D2 C7 EB 4E FD 9C 70 31 B1 11 8E D5 83 09 07 73 C6 16 FB CF 26 8C 24 5 [... cut ...]
[16:16:30][V][espdm:453]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 00 B5 14 82 E1 F5 15 C5 5B 34 A5 84 1E 3A 8B 77 70 BB D1 3E C6 69 EF 5E AF 62 EC 79 5D A4 2A F0 F4 7E B4 26 14 1D ED 07 1E 88 9C B6 C0 FC BD F8 9B EF 2E 8B CD 1D CE 7B 0C 17 48 CC E2 2D 4F 2F DD A1 1C 7F FB CD 93 BB E5 D7 B8 4A 3C DB FA BB 20 26 61 3C 27 FB 89 DB C7 F6 E7 C4 42 CF 31 53 AE 98 CB 30 25 C3 31 AE B7 2E 8F 01 53 08 E2 15 B3 A6 6F 66 FA 63 CF CD 9F 7A 65 D6 15 C3 61 84 0D 52 55 97 13 AA 9E 17 49 99 4A BF AD FA E1 42 21 3D 3D
[16:16:30][E][espdm:051]: Payload length is too big for received data

DomiStyle, 2021-09-19:
It seems like your data is getting cut off for some reason.

Can you make sure that the following line is present in your yaml? https://github.com/DomiStyle/esphome-dlms-meter/blob/master/meter01.example.yaml#L37

If yes, can you try increasing the read timeout to 200 and uploading again? https://github.com/DomiStyle/esphome-dlms-meter/blob/master/espdm.h#L34

color86, 2021-09-19:
hmm,

buffersize is set correct
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 2400
  rx_buffer_size: 1024
  invert: false
  id: mbus

i also set the read timeout to 200 but no change :(

DomiStyle, 2021-09-19:
Is it a Kaifa MA309M?
Also, what M-Bus to UART adapter are you using?

color86, 2021-09-19:
Yes a new Kaifa MA309M
I use a MIKROE-4137 UART adapter.

DomiStyle, 2021-09-20:
That's strange, do you have an USB to UART adapter available which you can connect to the Mikroe board and a PC to read out the raw data?

That would make it easier to see if the ESP is reading incorrect data or the Mikroe board.

color86, 2021-09-25:
Hello,
here the data with hterm from the ttl converter. 
[kaifa_raw.zip](https://github.com/DomiStyle/esphome-dlms-meter/files/7229890/kaifa_raw.zip)

how should i record the data or is it ok for you?

DomiStyle, 2021-09-26:
Yes, the raw data is ok for me.

For some reason there is no header at all in your data (68 FA FA 68), which is unlikely to happen in over 1 minute of capturing.
There is the header for the second packet though (68 72 72 68), so the baudrate seems to be correct.

Can you do another longer capture for like 10 minutes?

color86, 2021-09-26:
here is a longer capture
[output_2021-09-26_15-35-56_RAW.txt](https://github.com/DomiStyle/esphome-dlms-meter/files/7231465/output_2021-09-26_15-35-56_RAW.txt)
.

Exolor, 2021-09-30:
I am having the same problem with the tinetz kaifa MA309M.

I tried the possible fixes in this thread but no luck. 

I am using the ESP32 v2 Nodemcu with GPIO 4 & 36 like in the documentation.

Exolor, 2021-09-30:
Here is a snippet of the log if it helps, i can send you the decrypt key if you need it 

[14:05:15][D][espdm:040]: Handling packet
[14:05:15][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 51 31 C4 32 0A 71 BE C4 C8 54 48 32 D0 AF 31 93 44 C8 1E 19 9A F6 D1 A9 50 39 AF 71 95 D0 B5 22 33 37 91 2E FD 6B D1 08 C8 82 E5 73 E2 B1 B5 5A 04 94 C2 FC 46 45 7B DC B2 C6 70 CF 62 A6 8C FE DD CC 82 F6 2D BC 7A BC 7E 70 E8 1C D4 D6 33 93 1B B1 9F 97 8D 31 5B AF 32 91 EA 91 74 B0 EF 68 BA DE 8C FB 3F 4C 83 52 5D 90 F8 36 05 6E 6C 08 AB CF 89 6F D6 8F 9A 5D 0B 6C C1 15 07 2C 1D 13 F3 ED ED CE 50 DB F2 0C 28
[14:05:15][E][espdm:051]: Payload length is too big for received data
[14:05:20][D][espdm:040]: Handling packet
[14:05:20][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 52 23 F0 A6 B2 76 92 82 34 9D 6A 88 6E 22 95 E4 63 C1 AF AD FD 51 99 6D 96 9A 0C DA FD 2F 97 18 70 B9 74 BA 68 4B 9C 85 59 65 12 16 8E FF 05 E6 38 AC 41 E4 F0 DE 8A 7A 6C E7 25 F0 D6 D4 63 37 AD 49 CC CB 2E EC 68 F0 32 7B 83 F7 A4 3B 3C 2F 30 16 BA 76 0F 04 4A F7 E4 F1 6E 5E 1E FE B6 74 D0 A6 86 19 3B 4C 26 83 BB B8 D5 78 10 26 AC 0D A9 35 9B 82 82 6A 90 FB C8 8C 84 7F 77 E8 88 D5 2E 33 E4 58 6A A8 76 A8 25
[14:05:20][E][espdm:051]: Payload length is too big for received data
[14:05:25][D][espdm:040]: Handling packet
[14:05:25][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 53 D2 AE 15 F1 99 F9 83 42 41 8A 9D B3 4C F1 F0 41 47 A5 66 D3 DE B8 E3 B7 C6 71 15 39 33 6C 65 6F D7 99 06 0D 0B CF EC EF 0C 6F E2 65 3C D0 0D BE F6 05 B0 85 1E 11 8B 8D 33 98 89 31 77 F0 DE F2 A5 24 B2 A3 BE 83 70 D2 2C 4E 5D C6 82 CF 70 98 C6 9F 9E 93 ED C9 7E BF D4 0A CB 29 08 71 DC FB D1 26 8F 32 C5 05 65 F8 A8 D9 A7 51 04 AF 86 3B 47 95 F3 E2 55 33 61 4D 41 22 C7 78 65 0D CE EA B5 F0 B7 50 9C A1 E8 F0
[14:05:25][E][espdm:051]: Payload length is too big for received data
[14:05:30][D][espdm:040]: Handling packet
[14:05:30][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 54 D2 9C 54 2C 44 EF 25 85 0E C2 7B D2 06 5E 2B F9 B5 6E A2 B6 1D 75 61 89 BF 73 4F 3C 47 04 EC 1A 21 D3 7E 83 C9 F7 E4 AD C8 A1 A1 C1 EA 9B 09 25 6E 69 84 DE 93 2F D3 E3 FD 4A 66 19 90 2D D8 B4 E4 1B 24 84 A3 94 BB 5C B0 B6 14 75 6E 5F 6A 75 D1 6F 6F 31 F0 2A 9C CF 76 53 B1 43 8E B8 BF 8E 6B 33 CF 58 75 82 26 59 F0 29 D6 3B F0 4F A9 19 A3 5A DC E1 41 B2 96 B0 41 A8 AE B7 62 1B DE 74 77 27 97 FA 83 AE 32 2D
[14:05:30][E][espdm:051]: Payload length is too big for received data
[14:05:35][D][espdm:040]: Handling packet
[14:05:35][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 55 BF 56 CD 0D 7E 8D 16 F2 95 69 0E E5 C7 42 1C 06 FA 1C 59 E1 4E 83 10 BF D4 98 70 B3 09 77 33 C8 F0 94 DA FA 7B 4D 6D 22 B1 F9 1A 15 9B E3 79 B2 57 53 58 76 17 AD 59 49 6C B9 F7 F0 F3 42 87 DC 98 05 6B B1 82 4B 5B F0 91 C7 C1 A8 96 33 09 7D EA 7B 6E B0 04 3C 6D E1 72 C2 C5 5F 3F DA 05 38 3D 57 D5 07 30 92 FC DE 51 84 DA F9 21 25 33 C1 7F E3 5A 1D 89 62 2C 49 6F C7 2F 53 4A A4 CE 53 86 96 67 F6 5B 1D F0 7C
[14:05:35][E][espdm:051]: Payload length is too big for received data
[14:05:40][D][espdm:040]: Handling packet
[14:05:40][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 6D 56 70 30 17 B2 DB B0 37 FC 8A 96 E9 A9 3B B4 32 CA EE F0 68 38 ED EF DF 9F 38 F4 48 DC D9 9C ED FC 65 AD 15 42 21 56 5B 75 BB 86 8F 2A 30 F9 52 25 F0 A1 E4 CB 88 58 67 37 30 13 41 71 83 17 12 9E 8F ED 0A 71 74 52 66 66 AE 73 C9 C3 E7 F7 96 0D B9 86 E9 C8 DB 34 25 C3 A3 19 38 78 30 C3 DF F0 FA 70 66 58 3C 82 9D C7 F4 CB 17 39 77 16 F0 37 76 3A CC E9 CD 51 25 0D 2A F5 86 27 FD B4 B3 96 78 04 F7 76 F4 ED E5 0D

dbeinder, 2021-09-30:
I'm not using this project, but I'm playing around with a MA309 myself. The serial connection uses even parity, you could try adding this to the `uart` section in the yaml:
```
parity: EVEN
data_bits: 8
stop_bits: 1
```

Exolor, 2021-09-30:
Sure thing:

esphome:
  name: meter01
  platform: ESP32
  board: nodemcu-32s
  includes:
    - ./espdm

# Enable logging
logger:
  level: VERBOSE
  # tx_buffer_size: 2048 # Only needed when logging large packets

# Enable Home Assistant API if not using MQTT
# api:

wifi:
  ssid: xxx
  password: xxxxxxxx

ota:
  password: "xxx"

#web_server:
#  port: 80

uart:
  tx_pin: GPIO4
  rx_pin: GPIO36
  baud_rate: 2400
  rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  invert: false
  id: mbus

sensor:
  - platform: template
    id: meter01_voltage_l1
    name: meter01_voltage_l1
    unit_of_measurement: V
    accuracy_decimals: 1
  - platform: template
    id: meter01_voltage_l2
    name: meter01_voltage_l2
    unit_of_measurement: V
    accuracy_decimals: 1
  - platform: template
    id: meter01_voltage_l3
    name: meter01_voltage_l3
    unit_of_measurement: V
    accuracy_decimals: 1

  - platform: template
    id: meter01_current_l1
    name: meter01_current_l1
    unit_of_measurement: A
    accuracy_decimals: 2
  - platform: template
    id: meter01_current_l2
    name: meter01_current_l2
    unit_of_measurement: A
    accuracy_decimals: 2
  - platform: template
    id: meter01_current_l3
    name: meter01_current_l3
    unit_of_measurement: A
    accuracy_decimals: 2

  - platform: template
    id: meter01_active_power_plus
    name: meter01_active_power_plus
    unit_of_measurement: W
    accuracy_decimals: 0
  - platform: template
    id: meter01_active_power_minus
    name: meter01_active_power_minus
    unit_of_measurement: W
    accuracy_decimals: 0

  - platform: template
    id: meter01_active_energy_plus
    name: meter01_active_energy_plus
    unit_of_measurement: kWh
    accuracy_decimals: 0
  - platform: template
    id: meter01_active_energy_minus
    name: meter01_active_energy_minus
    unit_of_measurement: kWh
    accuracy_decimals: 0

  - platform: template
    id: meter01_reactive_energy_plus
    name: meter01_reactive_energy_plus
    unit_of_measurement: kWh
    accuracy_decimals: 0
  - platform: template
    id: meter01_reactive_energy_minus
    name: meter01_reactive_energy_minus
    unit_of_measurement: kWh
    accuracy_decimals: 0

text_sensor:
  - platform: template
    id: meter01_timestamp
    name: meter01_timestamp

mqtt:
  broker: 192.168.0.50
  id: mqtt_broker
  username: xxx
  password: xxx
  
 
  
custom_component:
  - lambda: |-
      auto dlms_meter = new DlmsMeter(id(mbus));
      byte key[] = {0x67, 0x45, 0x33, 0x62, 0x51, 0x42, 0x58, 0x42, 0x76, 0x6E, 0xxx, 0xxx, 0xxx, 0xxx, 0xxx, 0xxx};
      dlms_meter->set_key(key, 16);
      dlms_meter->set_voltage_sensors(id(meter01_voltage_l1), id(meter01_voltage_l2), id(meter01_voltage_l3));
      dlms_meter->set_current_sensors(id(meter01_current_l1), id(meter01_current_l2), id(meter01_current_l3));
      dlms_meter->set_active_power_sensors(id(meter01_active_power_plus), id(meter01_active_power_minus));
      dlms_meter->set_active_energy_sensors(id(meter01_active_energy_plus), id(meter01_active_energy_minus));
      dlms_meter->set_reactive_energy_sensors(id(meter01_reactive_energy_plus), id(meter01_reactive_energy_minus));
      dlms_meter->set_timestamp_sensor(id(meter01_timestamp));
      dlms_meter->enable_mqtt(id(mqtt_broker), "meter01/data");
      return {dlms_meter};

dbeinder, 2021-09-30:
68 FA FA 68 53 FF **F0 C1** 67 DB
Those two bytes from Exolor should be 00 01. 00=(CI byte, segment #0, not FIN), 01=(STSAP byte). So I do think there is some corruption of the serial data.

Exolor, 2021-09-30:
> I'm not using this project, but I'm playing around with a MA309 myself. The serial connection uses even parity, you could try adding this to the `uart` section in the yaml:
> 
> ```
> parity: EVEN
> data_bits: 8
> stop_bits: 1
> ```

I did try it with these settings but sadly no luck.

Exolor, 2021-09-30:
> 68 FA FA 68 53 FF **F0 C1** 67 DB Those two bytes from Exolor should be 00 01. 00=(CI byte, segment #0, not FIN), 01=(STSAP byte). So I do think there is some corruption of the serial data. Which MBUS/UART converter do you use that can strip the parity bit from a UART stream?

https://www.mikroe.com/m-bus-slave-click I am using this converter.

dbeinder, 2021-09-30:
Afaik the mikroe device just passes the parity bit through. Did the data change after the setting parity to even?

Exolor, 2021-09-30:
> Afaik the mikroe device just passes the parity bit through. Did the data change after the setting parity to even?

```
[17:23:45][D][espdm:040]: Handling packet
[17:23:45][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 76 9F D3 C1 7F 8D 56 4A 8D 24 F0 D4 41 90 5C BD 7A 0E 61 F4 CB 4A C3 BC 3D D4 AF 86 C2 61 89 C8 BC 69 E6 C2 DA CD 3B 17 55 45 F7 C6 BD D6 45 B1 19 B4 27 53 B0 8F 82 05 AA F0 5E BF 3D F6 BC 98 63 0B 4F 1F 17 E1 0B BC 30 43 FD F0 C1 87 7A D8 ED B2 79 99 0D F0 CB 2E 56 7E 1D 90 59 A6 2E 1E 7D 09 AF 70 AC 70 C8 A2 FF CB D1 55 8D A2 6D 29 44 75 CC C5 97 7C 6B A9 69 A8 31 E8 22 B9 0F 29 91 43 CB 17 19 3B 0C 82 30 F0
[17:23:45][E][espdm:051]: Payload length is too big for received data
[17:23:50][D][espdm:040]: Handling packet
[17:23:50][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 76 B0 1E 9D 68 BE B9 95 AA F0 BD F0 9B 2E 84 2F 50 7C DA DD 8F 23 B9 B7 BE DE 0B 3A 69 ED D7 4E 7E D8 A7 44 A6 57 97 30 1D F2 3F C7 39 C4 AD 69 95 82 13 D7 E8 C6 B2 BD 13 EE 10 58 8E 8D 46 30 D4 CE C3 70 1F A4 16 48 1A 58 F6 B9 DF 65 D6 51 3D 5D 2E 10 C2 61 2C E7 22 4D EF FE FC 2F 05 CF 99 B3 E8 58 5F B3 A8 4B 47 B6 B4 AE 49 E2 C1 55 B1 B0 8D E7 8F C9 1C 6F 49 ED AC 79 4A C4 92 71 B9 CE 9A B6 3F FF 58 19 BA B9
[17:23:50][E][espdm:051]: Payload length is too big for received data
[17:23:55][D][espdm:040]: Handling packet
[17:23:55][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 76 A1 FC B4 F4 70 AD E3 7C C1 CC D7 7F C4 13 DF F1 51 30 CC 69 72 AF 05 E7 F0 ED C8 F8 54 AC C4 C5 7D 09 95 F5 F5 53 A3 9C 6A 90 17 76 F1 F8 33 D8 EA F7 43 27 95 7B C4 72 05 0D C9 CE 3F 8B 77 89 18 B9 5A 23 DF 4A AA 1B E3 66 D5 9B 5C FE 4F 37 CE 0F EF 39 70 76 EE B1 F2 13 F0 8C 44 93 C8 28 0B 4F 7D 75 F3 8C 5D 41 82 67 92 99 E8 68 7D 67 D6 95 86 68 43 F2 A1 31 28 F2 68 29 54 C3 D1 46 F7 2B 61 86 05 48 8D 29 DA
[17:23:55][E][espdm:051]: Payload length is too big for received data
[17:24:00][D][espdm:040]: Handling packet
[17:24:00][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 76 A2 DC 61 26 78 FA 07 A2 14 17 5D 90 AE 2B AF 25 6E 2A 5D 85 5B 2F CE 3F D4 D2 C4 83 8F F0 AD 29 CB 74 3F C1 4E 8E BF C6 A1 B4 79 D0 10 E8 0B 51 75 49 8A 6D CB FF 2D 7B 3E 1B C7 82 F0 94 D3 CB 37 44 1D 6F 6D E7 E4 F0 B0 AB F7 69 47 E2 70 F0 B2 99 BF 4F 24 BB C1 B6 BD D9 2F 07 54 31 8B C1 34 EE C8 76 D5 EF 4F C5 BB 79 07 25 F6 A6 0C 17 D7 5D 86 65 8E EF BE BC A8 F0 64 38 67 A9 0F A5 91 70 97 16 70 21 6C 61 C6
[17:24:00][E][espdm:051]: Payload length is too big for received data
[17:24:05][D][espdm:040]: Handling packet
[17:24:05][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 76 A3 B6 EF 33 F9 A9 BB E2 15 B1 C2 70 4F F2 2E 0A 3E 44 6F 70 5A CD 39 3D 9E D8 DA 5E F6 8B D5 A2 D2 F0 4E 6E 07 D5 2C 8E 0B C1 51 7A BA 43 64 8A 08 CB 45 75 2D BC DB 7C B9 E5 30 BA BD 68 7B 70 DB CA 8A 45 AD C8 48 B2 FD 0C B7 78 C1 4A 97 6B D4 C4 09 B8 26 B3 9A 87 41 08 EC EB 30 DA 48 2D 4A 58 C8 71 89 1C 63 7E 5A 70 6B C2 0C 25 5C E4 6A F0 A4 30 E2 56 A6 3F 90 61 A1 50 30 EB A7 B6 82 84 D7 37 D8 9A 4B 1C 41
[17:24:05][E][espdm:051]: Payload length is too big for received data
```

This is the data with parity even, doesnt seem to change anything

Exolor, 2021-09-30:
So one thing i did test was setting the tx_buffer_size in the logger component, now i get longer packets but also "took a long time for an operation". 

```
[17:57:50][D][espdm:040]: Handling packet
[17:57:50][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 78 38 33 A3 B7 C4 DE 17 EC F2 42 D1 BC 82 6F B6 F7 67 9A 50 AB 32 F8 91 39 9B 1B F6 51 3B AC 73 15 5F CA 56 E4 3F 69 5D 75 78 3E 39 0E 69 A8 E4 AA D6 24 EA 05 C6 63 9D 5A 99 A3 69 12 71 E2 51 1D 6F F1 A9 16 83 4A 70 F4 0A 13 56 FC CE AC 49 46 A1 0B 14 7E 5A 45 E9 C1 6D CD E2 AB 32 64 9C 45 1F 26 94 95 05 79 4F F2 EE 49 0A 3B E2 39 7A 47 51 F5 57 BA 88 0E BC 12 39 13 9C 83 C1 A1 DF 94 04 D5 E1 0E A1 EE 67 2E EF A1 C3 D0 AD 96 D5 B9 B7 F8 D6 33 43 04 1E 83 CA CC E8 46 3F 6F 2E 2C B1 D5 71 BA FA 46 E3 13 53 B0 50 F7 C4 C9 39 2A E6 22 68 9F 11 85 3A 07 2D C8 3C C2 E3 9C D9 18 F8 33 37 9C 74 D0 09 4B DF 8A 5A 61 38 3F 62 4F 89 41 08 CE B3 91 8F ED 6B 42 66 1D FC 8A 48 61 4D 18 4E 44 F0 16 68 72 72 68 53 FF 11 C1 67 33 D0 F0 F7 D2 A4 95 88 42 2C 9B 4F AE F8 AF 27 CA 56 6C 38 FE 0F B0 75 47 68 8A C5 7F 19 05 F7 F1 96 4E F3 ED 37 D1 89 B5 4A 30 B9 84 1C 78 2D 7E D0 BF 61 AB 76 E1 65 E8 5C 28 97 F0 8E 5B CD 98 08 CD 34 70 7A 1C BA 9F 1D 88 50 59 9E B0 39 94 91 1C 11 6F B4 E8 2A 8D 3B D0 7F 7E E7 68 2C C6 F2 E6 BC 84 83 F4 B0 C1 33 B1 79 E3 57 16
[17:57:50][E][espdm:051]: Payload length is too big for received data
[17:57:50][V][component:207]: Component <unknown> took a long time for an operation (0.10 s).
[17:57:50][V][component:208]: Components should block for at most 20-30ms.
[17:57:55][D][espdm:040]: Handling packet
[17:57:55][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 78 39 98 D2 11 83 18 BA 70 FF 6F FE C2 CE A6 BF CC F7 0E 47 E9 B0 56 84 A3 61 CE 44 25 F0 79 63 3A 13 83 BC B5 E2 89 59 2F BB C7 39 D1 C1 85 36 DA BA 3C 9A 28 8D F9 94 0F C1 23 94 23 7F D8 B5 D1 FF 29 93 C8 F5 1F 51 A1 44 6C E1 FD D0 3D EA 29 51 C9 48 14 BF 55 B6 3F 9E AB 48 6B 6C FD 9E 08 1F 44 CF A5 DC 72 A6 D4 F7 86 D1 11 51 4B AE A9 33 75 BD 2E 5A D7 D9 9B 6C 56 F7 69 13 93 44 EC 18 72 B0 87 2C D4 8C 4B 2B 36 9A A4 BF 63 34 D0 BF 3F D8 8E EF B8 DA F0 82 26 84 24 AA 9C 3F 13 91 95 F5 2C 6B 49 A3 59 87 91 79 25 12 F5 A7 78 0B 17 0B C2 C1 6A BB FA 30 EB 5D 34 6C 2D 5C 0B 98 87 C7 D9 34 5E 71 30 AE CC 7C 25 A4 66 98 C2 45 2F 6E 79 A4 8D 86 8F 86 A3 52 08 C3 79 2D 8D 21 EE 4A D8 07 16 68 72 72 68 53 FF 11 C1 67 97 59 87 FB 3F 14 E3 9A D6 F0 87 37 65 76 05 12 0F 70 8B B1 8B E5 18 91 72 44 3E 2A F0 CB 0A 51 F0 92 23 61 6E 30 95 05 A9 0C 5C E1 48 C7 55 EF 77 5F E7 0F A7 B1 0A 1A F0 E1 63 3A F1 32 BC 77 E3 D1 93 3D 83 EE 1B 4E A8 DE BD 4D 55 E6 AF 24 4D EA F0 06 CE 79 A6 C1 56 FE 8A 93 35 63 89 FE 86 95 CB 2A B3 FB 70 9E 27 0C AB 0D 94 1C 16
[17:57:55][E][espdm:051]: Payload length is too big for received data
[17:57:55][V][component:207]: Component <unknown> took a long time for an operation (0.10 s).
[17:57:55][V][component:208]: Components should block for at most 20-30ms.
[17:58:00][D][espdm:040]: Handling packet
[17:58:00][V][espdm:453]: 68 FA FA 68 53 FF F0 C1 67 DB 08 4B 46 4D 10 30 C1 3E 56 82 C1 55 21 F0 05 78 3A 88 CE 6C D2 94 F6 06 EE 5E A5 FC 6D 94 32 EB 9E 0E B9 9F 34 DF 67 8D E9 5F 5B 70 0F F0 CE F5 92 D7 2A 8F AF 88 C7 47 F7 AD B0 D8 D3 C1 78 DE FC D6 6F 9C 87 BA 76 54 07 FF CB 90 FA BB F9 DB 9C A7 FE 30 C8 2B 06 83 1B C4 FC 23 52 A8 6D AB 5D C2 2E 64 D5 38 D3 F0 DD D7 BF 51 04 B4 65 30 76 8A 30 39 84 98 AF E9 97 45 33 5F B9 33 F2 83 BF C2 2F C3 76 5B 70 56 6F F5 6B 42 76 6F 15 19 98 41 FD 42 3F AC 7F 5B 10 47 6F 50 52 7B 1F 86 38 38 F5 82 A9 D6 33 16 F9 64 AE 86 C1 21 A2 F8 DC C2 1A 7F 33 A3 34 97 F1 4B E9 BA DE F2 28 AC 73 54 F9 E1 D9 FB 82 A8 4C 54 91 6A F1 70 82 DD 2B A8 07 74 B3 82 36 2A 8E C8 AE F0 77 8C E2 DE 3E AE CD DD D0 30 1D 37 89 91 28 82 70 EE 52 E3 3C CF F8 A2 2D 16 68 72 72 68 53 FF 11 C1 67 13 B9 D2 5A 18 93 C7 0D BD 07 8C E8 08 B6 50 F4 FF 93 DA 1F 04 FF 64 68 8C 21 8D A7 54 64 95 C1 70 94 B5 36 35 38 EE 8D 2B 8B 44 72 C4 DB 49 21 73 13 41 72 54 C1 FC 95 12 B3 10 FF D5 BF BE E4 78 F2 62 1D 97 AA B2 46 CF FC 52 F8 12 EF F8 78 28 B0 FC 36 B0 F9 F1 B0 BD A4 CA D5 E6 B5 61 B4 D4 5B A6 C4 B7 4F 83 1B 11 5D 19 90 B0 EA 16
[17:58:00][E][espdm:051]: Payload length is too big for received data
[17:58:00][V][component:207]: Component <unknown> took a long time for an operation (0.10 s).
[17:58:00][V][component:208]: Components should block for at most 20-30ms.
```

DomiStyle, 2021-09-30:
@Exolor Your packet looks complete now, yes.
Will take a look at your log once I get home.

dbeinder, 2021-09-30:
The data has the exact correct length, but there are bitflips in some bytes. One of those mangled bytes encodes the frame length, and decoding fails because the code now expects more data than is actually there.
Those flips seem to be very consistent (00 => F0, 01 => C1) that is why I suspect either the parity bit being read as data or there is an issue with the converter struggling with repeated 0 bits.
Unfortunately I'm not working with a ESP myself, so I can't help here other than saying the KSMWest format definitely is using even parity, not standard 8N1.

Exolor, 2021-10-01:
I can rule out the ESP32 for now, tested it with different board (d1 mini but esp32) and i get the excact same reading. 

So either its the meter, the converter or the software. I will rule out the converter later and report back.

Exolor, 2021-10-01:
Alright, so with a simple voltage divider 100k/10k i get the correct readings. 

So the converter was def the problem.

DomiStyle, 2021-10-01:
@Exoler Awesome, so in order to get things working you had to add the voltage divider and add `tx_buffer_size: 2048` to your config?

My converter is connected directly to an ESP32 with PoE, so there might be some hardware difference or interference.

dbeinder, 2021-10-01:
If my hunch about repeated 0 bits is correct, it may help to make the sampling capacitor larger (C2 on the mikroe board) https://www.ti.com/lit/ds/slas222b/slas222b.pdf

Exolor, 2021-10-01:
> @Exoler Awesome, so in order to get things working you had to add the voltage divider and add `tx_buffer_size: 2048` to your config?
> 
> My converter is connected directly to an ESP32 with PoE, so there might be some hardware difference or interference.

So right now i am using just 2 resistors between the mbus + and - without any other components.
[image]

It works with both `tx_buffer_size: 2048` on and off. 

It´s an hardware issue for sure, i will give the C2 a try like @dbeinder suggested, as i get the 68 FA FA 68 53 FF 00 01 packet now.

color86, 2021-10-03:
@DomiStyle 
I got it to work too. I have interrupted the 3,3V connection from esp32 to the MIKROE-4137 UART adapter
[image]

homecineplexx, 2021-12-03:
Könnt ihr mir bitte sagen, wie genau ich diese 100k und 10k Widerstände einbauen muß, damit das funktionieren könnte?
bei mir ist das Paket zu klein!