DomiStyle/esphome-dlms-meter #3: Payload length is too big for received data [TINETZ]

DaCapoAT, 2021-09-24:
Hi,
i also have the same issue like user color86 with too large packets.

Using ESP32 (nodemcu-32s) with microe-4137 m-bus slave.
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 2400
  rx_buffer_size: 1024 
  invert: false
  id: mbus
logger:
  level: DEBUG
  tx_buffer_size: 2048

readTimeout = 200

LOG:
Sometimes i get:
[E][espdm:036]: Received packet with invalid size

And then:
[D][espdm:041]: Handling packet
[D][espdm:454]: 37 32 FF B5 95 D1 95 C9 C1 31 2F D9 85 A5 B1 85 89 A5 FD 0D 0A 1B 5B 30 3B 33 B5 C9 81 27 6D 65 74 65 FE A8 B5 D5 C5 D1 D1 B9 73 65 6E 7B EB C1 A5 8D E9 20 27 FE EF B4 B4 15 DA 9A 8A 6A 6D D5 57 96 99 B1 BD DD 6D 5B 30 6D CD CD BD C9 81 27 6D 65 74 65 A8 B5 D5 C5 D1 D1 B9 73 65 6E 7B EB C1 A5 8D E9 20 27 FE FF B4 B4 05 DA 9A AA 6A 6D 8D C9 81 27 48 64 FE FE FE FD FD 56 E1 77 FF FD 96 CD FF FE FF 13 C8 F2 FF 96 9E 7D 1D E3 F3 FF 4D CF E8 9A DD BD 32 2E CE 14 AB B2 FF 2D 16 DB 55 B2 5B 0B 9C BF 1B 1D FD B0 CF F4 38 0D DB F6 C2 BA 1B 3F 21 68 BC 0A 7C 9D 15 11 F7 6B 94 9A FF FD E5 D1 BE 18 A9 4B 33 FC FE A0 C7 6D E2 C3 DC 3D D9 CD 43 E9 FD 3E 8E F0 B4 AA 2B 58 82 99 FE 77 13 16 25 E7 12 FF 9D DF 54 38 A9 26 DD E4 27 8E B4 96 6A 35 0A 1B 5B 30 FF 7D CC FE 77 3C FA FD 95 FF FF 33 B5 6A C4 3E 8D 4C E6 D9 E8 8C 3E B8 57 52 91 68 13 73 AD E6 CF 4B 24 F8 FE 4F 98 DC A7 5C 5F 7A 77 F7 5D 14 9B 0A 0B 38 49 AE CA 33 CF 59 3E 1A 34 F8 B9 C3 37 16 B9 8A FA 8D D5 C9 C9 95 BD FD FF B5 C5 D1 D1 B9 73 65 6E 73 6F 72 3A FF 6A 95 D1 95 C9 C1 31 B5 B5 C5 D1 D1 B9 73 65 6E 73 6F 72 3A FF 6B E9 20 27 6A 77 73 6D F1 AA 6A 6D 0D 75 6D B5 C5 D1 F1 BD 81 27 6D 65 74 65 72 FC A8 B5 D5 C5 D1 D1 B9 73 65 6E 7B EB C1 A5 8D E9 20 27 FE EF C9 7D C1 B1 D5 CD BD 73 74 61 7E 9A 95 B9 CD BD C9 81 27 6D FF A8 B5 D5 C5 D1 D1 B9 73 65 6E 7B EB C1 A5 8D E9 20 27 FE EF C9 7D B5 A5 B9 D5 CD BD 73 74 61 74 FF D2 55 B9 CD BD C9 81 27 7F 43 5D 5B 6D 71 74 74 2E 73 65 6F 7B EB C1 A5 8D E9 20 27 FE EF FF 9D E5 7D C1 B1 D5 CD BD 73 74 61 7E 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 90 FF FD FA FF 39 6F A9 F5 76 7D 78 23 31 23 29 6A BF 4C BF 52 1C BC AE 5A A3 2C A5 C3 35 D4 9B 96 68 7A FD DF F8 FF AA D4 65 7A 62 1B F9 FF 9B AE FF E9 F2 35 F3 6F 43 FF 5B 7E 7C D0 FE FD FF 2F A6 37 38 D6 D7 B6 87 18 BB B5 01 9E FF 53 D1 67 77 DC CB A8 8A E8 2A 68 39 D7 5C DB 97 37 6F D4 16 43 A0 F9 73 E2 D7 59 FF 8A 7A 9A 95 B9 CD BD 09 99 FF F5 47 DE EC 82 4D FE B0 CB 6A F2 FC 36 1D 3C 6C C3 75 8E D1 BE 7F E8 EA 3F F8 FF 88 95 B9 CD BD C9 E9 30 32 FF 52
[E][espdm:052]: Payload length is too big for received data
[D][espdm:041]: Handling packet
[D][espdm:454]: 1B 5B 30 3B 33 36 6D 5B 44 5D 5B 65 73 70 F4 95 D1 6D 5B 30 6D 0D EA 81 42 35 20 FF 32 46 20 44 39 FF D5 20 46 44 F8 00 33 42 20 FB B2 AA 02 BA A2 C2 55 02 22 C5 FC 05 2A 09 81 02 32 15 81 FF 0A 81 38 41 FF 39 39 20 42 31 FE 02 B2 22 81 FB 32 37 20 36 44 FC 1A D5 20 44 B9 81 37 42 20 E5 02 92 BA 02 32 0A CA 0A 81 8A 02 92 2A FF E5 F7 6C FF 35 9E 08 AF 60 77 5C 5F E8 C5 79 FD EF 5A 86 D1 7D AB 21 3A E9 FF FF FF 7F 42 33 85 6D E3 3D B8 B1 A3 F0 65 FC FA FF F7 01 73 F7 E9 17 C2 93 1B FF E9 FF 8A CF CB FC E5 20 43 44 F8 F5 57 77 F9 C1 7D CF 5C 5A 39 54 D4 D4 6B B5 7E EC 89 A1 4C 01 F7 DC 52 AF 13 9E B1 17 D6 10 D7 81 97 73 E1 05 7F E7 6C 73 73 CF FD 30 92 20 53 FF 3D 82 9F F8 D2 FD 8A 32 FF DF 39 EA 73 B9 16 14 CA AA 02 B2 0A C2 22 C1 05 12 E1 A0 00 81 45 36 F8 32 81 39 38 F0 02 32 DD 20 35 FF 02 A2 CA 02 EA 05 8A 0A 81 02 12 E5 20 FF 95 02 CA AA 02 FE 44 31 20 44 FD 05 BA 92 02 9A EA 02 1A E5 20 F3 02 22 C5 A0 32 20 33 41 F0 05 BA BA 02 BA 9A E2 C1 44 20 37 35 FE 42 44 20 38 F9 02 32 0D 81 FF 01 12 E5 20 37 35 20 38 44 F0 02 BA 22 81 F3 01 BA A2 02 B2 EA 00 1A E5 20 FF 81 44 35 20 FB D5 20 36 45 A0 E5 20 32 30 F8 13 0A D5 20 34 20 36 31 E0 00 1A E5 20 FF C9 22 81 37 31 FE 05 BA 12 81 F5 01 92 BA 02 32 81 43 31 20 FF 02 B2 8A 02 BA EA 81 46 46 20 FF 68 FA FA 68 53 FF 00 E1 B6 5B FE FF 1D 35 F5 FF F3 E1 5F 55 AA F5 87 F4 1E B3 52 31 BF 64 CF 3A FC E4 30 2D FE 5B 2E FE 67 F8 F5 7E B2 99 FF AF E5 AE 54 AA 38 2B 4B BC BD 51 8E B8 7F 1D 87 A7 E4 C8 CE 66 51 DE 5C DD C6 E5 6A 4E DD 2A DC CB 13 A3 44 2E 7F F4 AC 5E 4B DC C2 53 A0 F4 43 A7 0B 55 CB 85 0E DF 0D 51 E6 EA 26 8E 3D 12 B8 3C FE FF D7 FE FE 2F D2 BB C3 E6 66 17 CE 4E B5 16 24 00 E0 66 B5 A8 F4 ED 11 38 F6 3D 10 FC FA FF 6F 5D EA F2 F2 48 41 F2 EA FF 64 DE 73 95 3F 32 91 2A F4 8D FF BE 9C 8E AE A4 FD FD 7B E9 A9 6E 4C 16 35 A1 8B 9F 47 0D B3 5A 38 7A B3 88 F4 F9 45 38 20 45 E1 42 39 20 43 E4 00 46 46 20 B5 91 B5 E9 30 35 32 5D BE A5 9D 81 66 6F 72 20 FF
[E][espdm:052]: Payload length is too big for received data
[D][espdm:041]: Handling packet
[D][espdm:454]: 1B 5B 30 3B 33 36 6D 5B 44 5D 5B 65 73 70 F4 95 D1 6D 5B 30 6D 0D EA 02 9A 12 81 FF 90 4D 28 B2 FE 22 81 35 42 FE A2 92 02 9A AA F2 13 13 9A CA 02 FF 85 02 82 82 FA 2A 81 30 32 02 92 92 02 DA 02 82 92 02 FF 02 9A C2 02 F2 8A 02 32 15 F1 DD 20 32 30 F0 02 A2 A2 02 FA 02 CA 92 02 D2 8A 02 C2 0A F9 32 DD 20 36 43 FC 02 BA BA 02 AA 1A 45 46 20 35 41 0A 2A E5 20 FF 02 C2 AA 02 B2 FE 02 32 C1 A0 01 BA 9A 02 32 DD 02 32 19 81 FF 30 20 34 33 F0 77 69 7F 9D 1E FF D7 68 2B 15 CF FA 47 59 69 FC 6D 72 59 62 57 FE F7 E2 C7 60 CD BF 0D 51 0E A6 A1 9C E9 FF 3B EB 47 13 DB E4 84 D5 97 1D B9 43 DF 48 E8 0E 4F F0 47 E3 BB DF 06 87 77 23 F8 3E 3B 08 B9 A6 22 F2 FF E9 FF F4 DB A9 D2 E7 87 FC BF 26 44 22 FF F7 7F B7 BF A3 4B 37 20 71 35 FF 64 BA 1A 59 CA EF FF 7F EF C3 E9 E0 FF 69 16 F2 4D 9A 37 A9 A5 E2 7B 65 FC 20 45 41 20 FF FF 75 4A 8A FC 7F EA FD 93 37 56 FE 1F B3 35 29 A9 2C DF 52 CA F5 77 FC A7 BC C6 8A 5C 93 9A CF DF FF A7 C5 DA FD 9B B4 39 FD 09 C7 E7 56 E4 D4 1D 7A 98 63 62 AD 54 59 5E B9 DD 29 C6 0F 0D 1C DD 8C 35 CA 02 A2 A2 02 FA 32 30 20 33 38 FF 94 14 82 8A FA 82 02 9A C2 C2 C2 8A 02 32 CD FF 02 2A 05 81 38 31 20 34 34 FF 33 36 20 34 35 F8 02 8A 9A 02 F2 B2 02 9A 8A 02 FF 32 19 81 43 39 FF 02 12 05 E1 0A 82 92 82 05 82 92 02 FF C2 8A 02 A2 B2 FA 92 54 81 36 FE A9 12 81 46 F5 81 46 33 20 E5 32 D1 20 31 E5 20 81 33 41 F8 12 81 32 45 FE 02 CA CA 02 32 81 33 38 20 B2 45 20 42 38 FC 02 1A 15 81 02 2A D5 20 B6 25 81 31 33 FC 43 20 35 45 E0 46 34 20 34 33 FE 00 22 19 C1 00 9A 22 81 02 32 15 81 FF 02 B2 B2 02 8A BF FF 01 2A C1 A0 8A 02 9A C2 82 19 81 36 46 F0 02 32 C9 20 FF 31 4D FD FB 4E 9A D7 1F CF 51 5B 81 37 49 D5 B8 77 76 24 F2 FE 7D D7 A1 89 B6 B3 F3 45 56 2C FA 5F 25 F1 CC 26 57 DE 19 91 71 C4 56 F4 EF FE FD E5 F7 86 A4 5C 89 4B 56 53 BF BF 71 55 AD F6 93 FE D1 9C 66 48 C5 FF FD 45 DE EF 05 8C 8F B9 E9 03 A9 93 53 6F 62 29 BD C7 84 FF 60 1E 57 57 0B E3 79 8D 8A BB 2F 56 A3 78 64 0F 4A 47 D7 BC AF 21 F4 FE D7 F8 68 72 72 68 53 FF 11 01 67 85 C1 EF 00 76 68 C9 F2 CB 4C 64 A8 A5 E3 76 96 C5 4B FE
[E][espdm:052]: Payload length is too big for received data
[D][espdm:041]: Handling packet
[D][espdm:454]: 1B 5B 30 3B 33 36 6D 5B 44 5D 5B 65 73 70 F4 95 D1 6D 5B 30 6D 0D EA 02 9A 12 81 FF 90 4D 28 B2 FE 22 81 35 42 FE CA 0A 81 31 B2 90 28 26 B2 05 0A C9 A0 33 20 31 33 20 FF 32 20 38 32 F0 C9 20 39 32 F0 C9 20 30 32 FC C9 20 38 41 FC 01 9A 92 02 9A FF 02 32 05 81 05 82 92 02 FF 32 30 20 33 36 FE 82 92 02 0A FD 02 A2 8A 02 FE 30 32 20 43 B2 02 9A 92 02 9A 92 02 9A 92 82 90 94 14 9A FA 02 B2 CA 02 BA 32 32 42 20 31 35 FF C9 0D 81 36 E4 02 2A C9 20 FB 02 82 2A 81 14 2A 09 E1 35 20 39 37 20 B1 81 30 45 A0 81 30 36 20 FE C2 02 12 E5 FC 0A 32 D1 A0 9A E7 2E 79 F3 A8 5F 51 E3 71 2E 9F 5A CC 9F AF E1 7D 39 83 77 BD 3E 1B C2 D7 8D 88 77 F0 FF C7 D3 F7 D1 95 FF FC 5B FA 04 0D CB FD 65 DF FD D6 FA 57 A4 AF D7 33 66 95 67 6E 4E 99 E9 33 1E 26 29 95 F8 FF BF CD 1F 9B FF 4E 7F 5B BC CF 77 D8 32 C3 60 0D 23 A8 68 AF C3 FA 25 68 5D C2 CA F8 F4 F5 FF 95 EE 6F B6 77 2E 2D F9 95 64 69 3A F5 E6 D7 FF DE 27 D7 81 35 36 20 C5 5A F7 D4 FE 75 15 34 3B 77 61 88 FD 84 E3 B3 F4 D1 B3 41 4E E2 D1 FF FF EB 23 34 CF A9 35 75 74 9E D7 32 20 43 44 FC C2 02 9A 8A 82 02 92 82 02 FF 81 30 32 20 19 81 33 32 E0 02 8A 92 02 FE 92 02 82 AA E2 0A 81 30 32 FF 81 38 31 20 B3 81 46 35 20 FE 2A 02 92 82 FA 8A 02 32 E1 F8 00 1A 05 C1 80 20 42 32 FF 02 8A 0A 81 65 02 92 AA 02 FF 33 35 20 34 35 F8 01 32 15 81 05 92 92 02 FA 05 82 92 02 FF 32 19 81 30 FD 02 CA 0A 81 FB 02 32 C1 A0 02 A2 22 81 29 32 81 43 E6 D5 20 42 38 F8 44 37 20 41 31 FF D5 36 20 32 43 FE 01 AA BA 02 22 45 46 20 46 45 81 38 39 20 BC 35 20 41 44 FC 02 A2 C2 02 DA 14 82 AA 02 FF 02 CA 9A 02 AA DA A2 02 32 19 F1 BD 77 88 FF 7B 25 6E 54 A5 FE B8 B7 D0 3F 3B 3D B8 B7 72 98 C8 1F E0 C5 E2 FC 55 FD 73 A3 12 FE FF 5D 8B FF CD BA 15 87 77 69 FC FA F4 67 BF 1A C7 23 71 24 6D 63 FB E5 51 31 E9 69 C0 A7 93 F1 29 34 E1  [... cut ...]  92 CA 02 AA 25 02 92 8A E2 33 38 20 39 39 F0 AA 02 32 0D C1 0A CA A2 02 FA 53 02 C2 0A F9 35 33 20 46 45 FC 14 14 32 CD FE 46 20 35 37 20 B6 81 36 46 20 FA 05 9A 22 81 FF 0A C2 B2 02 EA 93 02 0A 09 90 92 02 EA FF 02 9A 92 02 9A 12 C5 20 39 B2 81 30 32 A0 39 20 41 30 FC 1A D5 20 32 F8 4A 02 82 92 F2 C9 15 81 30 D3 02 9A AA 02 FE 0A AA 12 81 0A B2 BA 02 BA E2 FF 90 54 02 FA B4 B4 15 DA 9A 8A 6A 6D 95 95 B9 9D D1 A1 81 69 7B FF 6A 35 0A
[E][espdm:052]: Payload length is too big for received data

DomiStyle, 2021-09-26:
The header is shifted in your packets (68 FA FA 68) or not present at all, which is strange.

Since splitting the packets is currently done by timing your smart meter might not adhere to that rule.
I will see if I can create a version which waits for the header instead soon, maybe that helps with your situation.

DaCapoAT, 2021-09-29:
Habe folgendes herausgefunden:
ESP32 nodemcu

"The logger component makes use of platform-specific hardware UARTs for serial logging. By default, the logger will occupy UART0. The ESP32 has three hardware UARTs, all of which can be used for both transmit and receive."

Standardeinstellung für TX/RX bei der NodeMCU ist:
tx_pin: GPIO1
rx_pin: GPIO3
Das entspricht der UART0 und daher kommt es zur Datenvermischung mit den Loggerdaten am UART-Port.

Der Logger kann auf einen anderen Port gelegt werden.
logger:
  hardware_uart: UART1 
  level: DEBUG

So, jetzt bekomme ich die Pakete mit 68 FA FA 68
ABER die ganze Nachricht ist nach wie vor nicht korrekt.
"Received packet with invalid size"
Daten sind zu kurz!?

[image]