DomiStyle/esphome-dlms-meter #6: always gets Errors

homecineplexx, 2021-11-30:
Hi
I tried this with different settings and always get this errors!
Maybe you can help me? would be great....thanks
greets chris

[image]

DomiStyle, 2021-12-01:
Can you post your folder structure, config and details about your ESPHome setup?

homecineplexx, 2021-12-02:
Here they are

[meter01.yaml (1).txt](https://github.com/DomiStyle/esphome-dlms-meter/files/7639459/meter01.yaml.1.txt)
[image]

DomiStyle, 2021-12-02:
Can you repeat the steps listed in the readme at https://github.com/DomiStyle/esphome-dlms-meter#software-installation?

Most noteably, your directory structure should look like this:

* config
  * meter01.yaml
    * esphome-dlms-meter
      * Contents of this repository (espdm.h, ...)

Make sure to remove any stray files from this repo you have added in your root config directory.

Change the includes line to this afterwards:

```
  includes:
    - ./esphome-dlms-meter
```

Don't include the .h file directly.

homecineplexx, 2021-12-02:
ok i will try this in the evening and give you feedback!
thanks

homecineplexx, 2021-12-02:
so i tried this, but there is the next or other error
[image]
[image]

[meter01.yaml.txt](https://github.com/DomiStyle/esphome-dlms-meter/files/7644344/meter01.yaml.txt)

DomiStyle, 2021-12-02:
Can you try with the new version I just pushed? Don't forget to clean previous build files and edit your config:
`auto dlms_meter = new esphome::espdm::DlmsMeter(id(mbus));`

homecineplexx, 2021-12-02:
ok, thanks, you are incredible. with the new version the compile works and i can test if the esp32 works with my smartmeter

thanks thanks thanks

closed 2021-12-02 by homecineplexx