DomiStyle/esphome-dlms-meter #9: ESP32 NodeMCU Module WLAN WiFi from AZ-Delivery

KnallerRaphael, 2022-01-16:
Hello.
Does this work with this https://www.az-delivery.de/products/esp32-developmentboard?
In the readme it said that the GPIO36 from the esp32 need to be connected with TX from M-Bus Board, but this one does not have a GPIO36 (or G36). Does it work with another pin?

mitsubishievo99, 2022-02-19:
Yes, just use GPIO16

cococheaf, 2022-03-02:
Hi Domi, liebe Grüße aus Höchst!
Ein super Projekt, das hat mir schonmal sehr weitergeholfen -  danke dafür!
Eine Frage habe ich aber dennoch: bei der Validation des Codes im ESPHome wirft er mir folgendes aus:

```
INFO Reading configuration /config/esphome/smartmeter.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Configuration is valid!
```

Ist das ein Problem oder kann ich den GPIO4 dennoch problemlos verwenden?

Danke  und LG - André

DomiStyle, 2022-03-02:
@cococheaf Ist kein Problem, die Warnung kannst du ignorieren.

cococheaf, 2022-03-03:
> @cococheaf Ist kein Problem, die Warnung kannst du ignorieren.

Ich danke dir!

LG

andreasfurxer, 2022-07-09:
Hallo,
Liebe Grüße aus Dornbirn. Super Projekt! ich bin begeistert.
Versuche gerade das Ding auch zum laufen zu bringen.
Compilen hat funtkioniert! nun bekomme ich aber trotzdem keine Daten aus dem Zähler.
ESP32 ist über IP Adresse erreichbar - MQTT Verbindung wird hergestellt.

Ich verwende diesen ESP32 **AZDelivery ESP32 Dev Kit C V4 NodeMCU WLAN WiFi**
Welche Pins müssen bzw. können hier für RX/TX verwendet werden?
**https://m.media-amazon.com/images/I/714PSaVZQtL._SL1500_.jpg**

Ich habs jetzt mal mit GPIO 16/17 probiert.
Dann muss ich doch auch den Eintrag in der yaml dementsprechend ändern.

_uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 2400
  rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  id: mbus_

Kann es sein, dass der Code der VKW nicht mehr aktuell ist?
Habe den Code bereits im Jänner angefordert und bekommen.

LG andy

DomiStyle, 2022-07-14:
@andreasfurxer Der ESP32 hat mehrere UART-Schnittstellen. UART2 ist TX GPIO16 und RX GPIO17, du könntest also probieren RX und TX bei dir zu tauschen.

> Kann es sein, dass der Code der VKW nicht mehr aktuell ist?
Habe den Code bereits im Jänner angefordert und bekommen.

Du meinst den Schlüssel zum Entschlüsseln? Würde mich wundern wenn der sich ändert.

andreasfurxer, 2022-07-29:
TX und RX hab ich mal getauscht. Läuft aber trotzdem nicht.
"Received packet with invalid size" wird mir beim Start jedes mal angezeigt.

luusl, 2023-08-28:
Funktioniert das eigentlich auch mit einem ESP8266 der auch von ESPHome unterstüzt wird, oder braucht es zwingend einen ESP32?

lechndo, 2023-10-05:
Hey guys,

just found this project wanted to give it a try, but I'm not yet fully familiar with the ESP32 family or ESPhome.
So I thought I could just ask you instead of spending a lot of time figuring this out by myself...

I found an old "AZDelivery D1 Mini NodeMcu ESP8266" from a previous project in my workshop and was wondering if it would also work with this board...
https://m.media-amazon.com/images/I/71b9yM7dFlL._AC_SX679_.jpg

If so, which pins should I choose for the serial?

From what I have found GPIO4 would be TXD1 but it seems the RXT1 is not available on this board and the pins 7 & 8 (RXD2 and TXD2) seem to be used for the flash interface.

Could I configure some other pins and use some kind of UART emulation or maybe somehow use TXD0 and RXD0?

Thanks in advance for your help

daowa89, 2023-10-15:
@lechndo
I also want to use the same board like you. At the moment I am using GPIO 1 and 3, but its not working.
Not sure if the logger of esphome is the reason (deactivated with baudrate 0), or the wiring.
So which GPIOs are you using?

lechndo, 2023-10-19:
> @lechndo I also want to use the same board like you. At the moment I am using GPIO 1 and 3, but its not working. Not sure if the logger of esphome is the reason (deactivated with baudrate 0), or the wiring. So which GPIOs are you using?

Hi @daowa89,
sorry for the delayed answer, but a lot was going lately and I just forgot to answer... :man_facepalming: 

I finally have a setup that works.
I connected D1 to the adapter boards TX and D2 to the adapters RX pin.

In the config I entered the usual things according to my setup (WiFi, smart meter key, MQTT IP, etc.).
The most important thing was to set the correct board
```yaml
esp8266:
  board: d1_mini
```
and to configure the serial according to the chosen pins. 
```yaml
uart:
  tx_pin: D1
  rx_pin: D2
  baud_rate: 2400
  rx_buffer_size: 2048 # Needed to receive the large packets send by the smart meter
  id: mbus
```
The correct pin naming is important because the pin names _Dx_, labeled on the board are not the actual ESP _GPIOx_ and the short names also only work with the correct board.

I hope this will be helpful for someone :wink:

chrispakr, 2023-10-20:
> > @lechndo I also want to use the same board like you. At the moment I am using GPIO 1 and 3, but its not working. Not sure if the logger of esphome is the reason (deactivated with baudrate 0), or the wiring. So which GPIOs are you using?
> 
> Hi @daowa89, sorry for the delayed answer, but a lot was going lately and I just forgot to answer... 🤦‍♂️
> 
> I finally have a setup that works. I connected D1 to the adapter boards TX and D2 to the adapters RX pin.
> 
> In the config I entered the usual things according to my setup (WiFi, smart meter key, MQTT IP, etc.). The most important thing was to set the correct board
> 
> ```yaml
> esp8266:
>   board: d1_mini
> ```
> 
> and to configure the serial according to the chosen pins.
> 
> ```yaml
> uart:
>   tx_pin: D1
>   rx_pin: D2
>   baud_rate: 2400
>   rx_buffer_size: 2048 # Needed to receive the large packets send by the smart meter
>   id: mbus
> ```
> 
> The correct pin naming is important because the pin names _Dx_, labeled on the board are not the actual ESP _GPIOx_ and the short names also only work with the correct board. I'm not sure if the GPIO numbers are correct, but from the pinout picture in my last post it seems that D1 is actually GPIO20 and D2 is GPIO19, etc.
> 
> I hope this will be helpful for someone 😉

Hello daowa89,

thank you for your description. - So you use the software-serial for receiving data from the M-Bus slave.

Unfortunately you made a mistake in your description:

you wrote: 
D1 goes to the adapter board TX (so D1 is RX on the ESP).
D2 goes to adapter board RX (so D2 is TX on ESP)

so you have mixed up the pins in the config.
correct is:

```yaml
uart:
  tx_pin: D2
  rx_pin: D1
  baud_rate: 2400
  rx_buffer_size: 2048 # Needed to receive the large packets send by the smart meter
  id: mbus
```

after that everything works fine! - Thanks

daowa89, 2023-10-20:
Thank you all for the feedback! I will test it on the weekend.