DomiStyle/esphome-dlms-meter #10: ESP32 Install Error

theexit, 2022-01-24:
Seit dem Update vom ESPHome bekomme ich beim Installieren diese Fehlermeldung.

```
src/esphome-dlms-meter/espdm.cpp: In member function 'virtual void esphome::espdm::DlmsMeter::loop()':
src/esphome-dlms-meter/espdm.cpp:371:22: error: no matching function for call to 'esphome::mqtt::MQTTClientComponent::publish_json(const char*&, esphome::espdm::DlmsMeter::loop()::<lambda(ArduinoJson::JsonObject&)>)'
                     });
                      ^
In file included from src/esphome/components/mqtt/custom_mqtt_device.h:7:0,
                 from src/esphome.h:10,
                 from src/esphome-dlms-meter/espdm.h:1,
                 from src/esphome-dlms-meter/espdm.cpp:1:
src/esphome/components/mqtt/mqtt_client.h:207:8: note: candidate: bool esphome::mqtt::MQTTClientComponent::publish_json(const string&, const json_build_t&, uint8_t, bool)
   bool publish_json(const std::string &topic, const json::json_build_t &f, uint8_t qos = 0, bool retain = false);
        ^
src/esphome/components/mqtt/mqtt_client.h:207:8: note:   no known conversion for argument 2 from 'esphome::espdm::DlmsMeter::loop()::<lambda(ArduinoJson::JsonObject&)>' to 'const json_build_t& {aka const std::function<void(ArduinoJson6185_91::ObjectRef)>&}'
*** [/data/smartmeter/.pioenvs/smartmeter/src/esphome-dlms-meter/espdm.cpp.o] Error 1
```

Anscheinend gab es ein Update von MQTT.
[](https://github.com/esphome/esphome/pull/2844/files#)

Meine Programmierkenntnisse scheitern bei c++ allerdings.

closed 2022-01-26 by DomiStyle

tomm84, 2022-05-16:
Hallo DomiStyle, der Fehler tritt nun auf wenn ich versuche es zu installieren. wenn ich das & in Zeile 333 wieder hinzufüge geht es  zum Installieren. 
Gruss Thomas

theexit, 2022-05-16:
> Hallo DomiStyle, der Fehler tritt nun auf wenn ich versuche es zu installieren. wenn ich das & in Zeile 333 wieder hinzufüge geht es zum Installieren. Gruss Thomas

Welche ESP-Home Version verwendest du?