DomiStyle/esphome-dlms-meter #13: [Question] Does this also work with a SAIA M-Bus energy meter

Kurisutian, 2022-04-02:
Hi,
while searching for a solution to integrate my SAIA M-Bus energy meter ([https://sbc-support.com/en/search/?q=ale3b5f+mbus&tx_solr[filter][0]= 3](https://sbc-support.com/en/search/?q=ale3b5f+mbus&tx_solr%5Bfilter%5D%5B0%5D=)) into ESPHome I found this repo and I was wondering whether it will also work to get the SAIA energy meter running or what eventually needs to be adjusted to do so. So far I couldn't get it to work and the MBus slave device mentioned in the docs is the same I am using so I'm curious about how to get it working with your example eventually. Do you happen to know if this might work? Thanks a lot!

DomiStyle, 2022-04-04:
If your packet starts with `68 FA FA 68` it may be supported but it's unlikely since every provider has to do their own thing.

Kurisutian, 2022-04-05:
Thanks for the information. Stupid question as this is my first encounter with something that was not working out of the box: What do I need to do to just get the packet(s)? I did not yet find out how to simply pass anything to the logs and be able to get further information from there. I checked the manual(s) from the link I posted but how do I actually parse the sensor data so I can compare or kinda use them? 🙈

bzed, 2023-03-15:
From my own experience of implementing a smart meter gateway prototype for one of the companies mentioned in this github repository: the mbus implementation always seemed to be well done. when you manage to decrypt the data, its usually no issue to parse it anymore - but different meters use different sensor types/ids or worse, their own  idea of how the data should be represented (endianess, data type, .....)

https://m-bus.com/ or the documentation of the meter is helpful there. Supporting other meters should not be hard, but it might need some coding to support new/other sensors.