DomiStyle/esphome-dlms-meter #14: [QUESTION] - Which data is available in general?

cococheaf, 2022-04-04:
Hi, 

everything is working as desired. Thanks for your great support.
I have a question anyway, what data is available in general, e.x. power on each phase, etc.?

Thanks for your help.

André

DomiStyle, 2022-04-04:
It's listed in the [README](https://github.com/DomiStyle/esphome-dlms-meter#exposed-sensors):

```
    Voltage L1
    Voltage L2
    Voltage L3
    Amperage L1
    Amperage L2
    Amperage L3
    Active Power Plus
    Active Power Minus
    Active Energy Plus
    Active Energy Minus
    Reactive Energy Plus
    Reactive Energy Minus
```

cococheaf, 2022-04-04:
Thank you! And how do i disable mqtt?

DomiStyle, 2022-04-04:
Remove the following sections from your config:

```
mqtt:
  broker: 192.168.1.1
  id: mqtt_broker
```

and

```
dlms_meter->enable_mqtt(id(mqtt_broker), "meter01/data"); # Enable grouped together MQTT report, useful to get exact time with each data for storing results in InfluxDB
```

Please note that without MQTT your data will not be grouped together so all sensors are updated individually at the time they are processed.

cococheaf, 2022-04-04:
Sadly, no - doesn't let me upload the firmware then:

Processing smartmeter (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome-dlms-meter/espdm.cpp.o
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome/components/api/api_connection.cpp.o
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome-dlms-meter/espdm.cpp:1:0:
src/esphome-dlms-meter/espdm.h:27:34: error: 'mqtt' has not been declared
                 void enable_mqtt(mqtt::MQTTClientComponent *mqtt_client, const char *topic);
                                  ^
src/esphome-dlms-meter/espdm.h:27:60: error: expected ',' or '...' before '*' token
                 void enable_mqtt(mqtt::MQTTClientComponent *mqtt_client, const char *topic);
                                                            ^
src/esphome-dlms-meter/espdm.h:64:17: error: 'mqtt' does not name a type
                 mqtt::MQTTClientComponent *mqtt_client = NULL;
                 ^
src/esphome-dlms-meter/espdm.cpp: In member function 'virtual void esphome::espdm::DlmsMeter::loop()':
src/esphome-dlms-meter/espdm.cpp:331:26: error: 'class esphome::espdm::DlmsMeter' has no member named 'mqtt_client'
                 if(this->mqtt_client != NULL)
                          ^
src/esphome-dlms-meter/espdm.cpp:333:27: error: 'class esphome::espdm::DlmsMeter' has no member named 'mqtt_client'
                     this->mqtt_client->publish_json(topic, [=](JsonObject root)
                           ^
src/esphome-dlms-meter/espdm.cpp:333:64: error: 'JsonObject' has not been declared
                     this->mqtt_client->publish_json(topic, [=](JsonObject root)
                                                                ^
src/esphome-dlms-meter/espdm.cpp: In lambda function:
src/esphome-dlms-meter/espdm.cpp:337:48: error: assignment of read-only location '"voltage_l1"[root]'
                             root["voltage_l1"] = this->voltage_l1->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:338:48: error: assignment of read-only location '"voltage_l2"[root]'
                             root["voltage_l2"] = this->voltage_l2->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:339:48: error: assignment of read-only location '"voltage_l3"[root]'
                             root["voltage_l3"] = this->voltage_l3->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:344:48: error: assignment of read-only location '"current_l1"[root]'
                             root["current_l1"] = this->current_l1->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:345:48: error: assignment of read-only location '"current_l2"[root]'
                             root["current_l2"] = this->current_l2->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:346:48: error: assignment of read-only location '"current_l3"[root]'
                             root["current_l3"] = this->current_l3->state;
                                                ^
src/esphome-dlms-meter/espdm.cpp:351:55: error: assignment of read-only location '"active_power_plus"[root]'
                             root["active_power_plus"] = this->active_power_plus->state;
                                                       ^
src/esphome-dlms-meter/espdm.cpp:352:56: error: assignment of read-only location '"active_power_minus"[root]'
                             root["active_power_minus"] = this->active_power_minus->state;
                                                        ^
src/esphome-dlms-meter/espdm.cpp:357:56: error: assignment of read-only location '"active_energy_plus"[root]'
                             root["active_energy_plus"] = this->active_energy_plus->state;
                                                        ^
src/esphome-dlms-meter/espdm.cpp:358:57: error: assignment of read-only location '"active_energy_minus"[root]'
                             root["active_energy_minus"] = this->active_energy_minus->state;
                                                         ^
src/esphome-dlms-meter/espdm.cpp:363:58: error: assignment of read-only location '"reactive_energy_plus"[root]'
                             root["reactive_energy_plus"] = this->reactive_energy_plus->state;
                                                          ^
src/esphome-dlms-meter/espdm.cpp:364:59: error: assignment of read-only location '"reactive_energy_minus"[root]'
                             root["reactive_energy_minus"] = this->reactive_energy_minus->state;
                                                           ^
src/esphome-dlms-meter/espdm.cpp:369:47: error: assignment of read-only location '"timestamp"[root]'
                             root["timestamp"] = this->timestamp->state;
                                               ^
[... cut ...]
src/esphome-dlms-meter/espdm.cpp:434:77: error: expected primary-expression before 'const'
         void DlmsMeter::enable_mqtt(mqtt::MQTTClientComponent *mqtt_client, const char *topic)
                                                                             ^
Compiling /data/smartmeter/.pioenvs/smartmeter/src/esphome/components/api/api_server.cpp.o
*** [/data/smartmeter/.pioenvs/smartmeter/src/esphome-dlms-meter/espdm.cpp.o] Error 1
========================== [FAILED] Took 3.01 seconds ==========================

cococheaf, 2022-04-04:
Nope - as broker is needed for completed syntax...

cococheaf, 2022-04-05:
if i just add:

mqtt:
  broker: ''

the it works but is throwing logs that it cannot reach the broker (logischerweise :)).

Ideal would be an option do disable either mqtt or the esphome sensors in general as both are created...