DomiStyle/esphome-dlms-meter #16: ISSUE: abnormal data in home assistant

cococheaf, 2022-04-06:
Hi,

i have found another issue, not sure if its the code, but I have the following problem:
I have now implemented the meter in home assistant and added it to the energy dashboard.
The problem now is that the meter is reporting data correctly, until once a completely malformed data is being sent from the meter.

See pictures attached:

[image]
[image]
[image]
[image]

How does it behave at yours?
I have activated the quarter-hour setting at Vorarlbergnetz. Just to be precise.

PS: Sorry, wenn ich auf deutsch schreiben soll einfach sagen :9
Und sorry wenn ich dich damit nerve, ich bin nur sehr interessiert an dem Ding.
Würde mich interessieren ob der Fehler bei dir auch auftritt.

Thank you for your support.

Andre

cococheaf, 2022-04-06:
Info: Found a similar problem here:
https://github.com/home-assistant/core/issues/57551

DomiStyle, 2022-04-06:
I had this happen once last year, unfortunately I didn't log the values sent by the smart meter at the time it happened so I'm not sure if it's the smart meter that sent invalid data or if the value of the smart meter rolled over somehow.

If it happens again I will add a limiter to the values the smart meter can send and ignore unrealistic values.

Unfortunately deleting values from the Home Assistant energy database is also a pain.

cococheaf, 2022-04-06:
Thanks to home assistant 2022.4 you can now easily edit the malformed value directly from dev tools.
Will check if the above issue is repeating and report back. sadly I can't code :(

cococheaf, 2022-04-07:
sadly again :(

[image]

Is it okay for you to apply the limit?

Thank you - LG aus Höchst

cococheaf, 2022-04-14:
So... i have now gathered more logs, seeing that the smartmeter is pushing wrong values from time to time.
Here's an excerpt from the logs:

`[20:10:30][D][espdm:042]: Handling packet
[20:10:30][V][espdm:455]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 01 D9 16 82 01 55 21 00 01 17 94 52 DC 7D C5 7F 68 AB 36 34 98 21 4A F7 A1 7D 6C C3 83 FE 47 BE E6 6F C4 81 B4 95 76 91 7D 66 E5 72 01 F3 EA 53 41 66 8F E4 33 47 F9 8E 73 33 92 05 13 E2 66 EC E1 17 90 3C 90 A6 9B 1C B3 C4 5F 48 59 4D 97 75 85 80 CC 64 FF 2F 87 82 4B B6 41 F4 C8 A1 F7 7E E4 00 30 52 E5 A5 B4 CE 8A 48 59 B1 1C 54 BC B7 7B 99 83 89 FA 1B 2F 19 E7 11 15 9B 96 D4 B3 EE 8C 99 62 8F AE 6D 13 23 74 14 70 2D 8C 68 57 21 FE AE F0 C4 AA 87 B3 65 1E BF BA 9F A3 5A 57 CE 24 F5 11 69 14 BD 84 AB 51 19 A0 C2 B3 3F 48 1D 95 A4 A2 07 0E 9A B6 52 49 C3 E5 79 A7 64 F6 D9 C5 FD 03 97 9C 84 06 63 9F 68 3C AA DA EC 71 27 DB 6B 50 3E 09 49 E6 9F 23 F7 0D 0A 15 E1 3C A4 6B DF 3A D7 4D 8B 05 91 2A 74 60 2F 5A 65 EB 16 68 72 72 68 53 FF 11 01 67 E1 E8 B2 4F 79 89 25 83 83 FB B7 5A 52 BD 26 60 3B 6B 32 D0 A1 C8 4B 13 8C 92 49 1A 45 9A DE 7A 50 7B AC E1 71 AA 4E 65 8B A5 03 48 89 27 13 0E 07 94 30 5B BD 91 B4 2B 74 D1 48 3C 93 FD 48 FD AE 26 6F 80 51 5F 53 0F 68 91 05 CE 5E 8D BB A2 5E C2 5D F1 13 91 32 A7 E5 B4 F9 CA 1B E9 88 8F AB 82 44 B1 58 12 AF E6 BB 99 53 66 F0 56 16
[20:10:30][V][text_sensor:016]: 'smartmeter_timestamp': Received new state 2022-04-13T20:10:30Z
[20:10:30][D][text_sensor:067]: 'smartmeter_timestamp': Sending state '2022-04-13T20:10:30Z'
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_timestamp/state' payload='2022-04-13T20:10:30Z' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_voltage_l1': Received new state 225.899994
[20:10:30][D][sensor:125]: 'smartmeter_voltage_l1': Sending state 225.89999 V with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_voltage_l1/state' payload='225.9' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_voltage_l2': Received new state 228.899994
[20:10:30][D][sensor:125]: 'smartmeter_voltage_l2': Sending state 228.89999 V with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_voltage_l2/state' payload='228.9' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_voltage_l3': Received new state 227.600006
[20:10:30][D][sensor:125]: 'smartmeter_voltage_l3': Sending state 227.60001 V with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_voltage_l3/state' payload='227.6' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_current_l1': Received new state 12.020000
[20:10:30][D][sensor:125]: 'smartmeter_current_l1': Sending state 12.02000 A with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_current_l1/state' payload='12.0' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_active_power_plus': Received new state 3196.000000
[20:10:30][D][sensor:125]: 'smartmeter_active_power_plus': Sending state 3196.00000 W with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_active_power_plus/state' payload='3196.0' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_active_energy_plus': Received new state 9757245.000000
[20:10:30][D][sensor:125]: 'smartmeter_active_energy_plus': Sending state 9757.24512 kWh with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_active_energy_plus/state' payload='9757.2' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_reactive_energy_plus': Received new state 51420.000000
[20:10:30][D][sensor:125]: 'smartmeter_reactive_energy_plus': Sending state 51.42000 kWh with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_reactive_energy_plus/state' payload='51.4' retain=1)
[20:10:30][V][sensor:074]: 'smartmeter_reactive_energy_minus': Received new state 2558061.000000
[20:10:30][D][sensor:125]: 'smartmeter_reactive_energy_minus': Sending state 2558.06104 kWh with 1 decimals of accuracy
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/sensor/smartmeter_reactive_energy_minus/state' payload='2558.1' retain=1)
[20:10:30][I][espdm:329]: Received valid data
[20:10:30][V][mqtt:405]: Publish(topic='smartmeter/data' payload='{"voltage_l1":225.9,"voltage_l2":228.9,"voltage_l3":227.6,"current_l1":12.02,"current_l2":2.93,"current_l3":0.27,"active_power_plus":3196,"active_power_minus":0,"active_energy_plus":9757.245,"active_energy_minus":0,"reactive_energy_plus":51.42,"reactive_energy_minus":2558.061,"timestamp":"2022-04-13T20:10:30Z"}' retain=0)
[20:10:30][V][component:199]: Component <unknown> took a long time for an operation (0.39 s).
[20:10:30][V][component:200]: Components should block for at most 20-30ms.
[20:10:45][D][espdm:042]: Handling packet
[20:10:45][V][espdm:455]: 68 FA FA 68 53 FF 00 01 67 DB 08 4B 46 4D 10 20 01 D9 16 82 01 55 21 00 01 17 95 FB CB 82 FA D7 10 88 72 72 25 2C 8E E7 39 B2 CE 95 0D 4A E2 A9 32 B7 58 C1 B7 28 1F 57 F1 D8 B9 20 52 D0 46 91 F5 78 E2 3E 92 9A 42 BF B4 23 10 FB E0 8D 5C 89 20 99 8C C2 0A 00 A8 DB F1 27 2C FD 44 DD 52 CB 84 01 D9 36 ED 8E AB 15 09 00 FC 17 99 DF 17 47 48 91 25 80 53 69 8A 43 60 D9 73 CC B4 69 AF 12 54 60 CE B9 35 A1 10 6F 49 B2 30 BE 41 1E 76 60 D6 22 49 55 56 BA 91 72 4C 6E 07 FE 54 FB C4 C4 B3 D7 CA 48 15 20 C6 36 01 2A F4 45 DE 2B DB A9 4E 5F CF FF E8 02 FB 78 B3 C3 BB E6 98 15 96 DD 66 99 FF C6 3F 93 9C 0C 1F B6 DF 18 A4 F2 AB BB 5B 2E 04 D4 52  [... cut ...]
      dlms_meter->set_active_power_sensors(id(smartmeter_active_power_plus), id(smartmeter_active_power_minus));
      dlms_meter->set_active_energy_sensors(id(smartmeter_active_energy_plus), id(smartmeter_active_energy_minus));
      dlms_meter->set_reactive_energy_sensors(id(smartmeter_reactive_energy_plus), id(smartmeter_reactive_energy_minus));
      dlms_meter->enable_mqtt(id(mqtt_broker), "smartmeter/data");
      dlms_meter->set_timestamp_sensor(id(smartmeter_timestamp));
      
      return {dlms_meter};`

Maybe someone can see the problem.

DomiStyle, 2022-04-14:
In what intervals does this usually happen?

cococheaf, 2022-04-14:
unregularly - there is no specific time when that happens. i had it once at the 12th of April at 6pm, the once at 11:20pm and again yesterday at 08:10pm. i cannot find any issue with esphome itself or with the lambda calculations. your code must be correct as it gets values from the smartmeter, so i assume there is a problem with either the smartmeter or the MBUS Slave Click.

The system is wired as you suggested. +3v3 is coming from the esp32 chip. the rj11 cable is shielded.

cococheaf, 2022-04-14:
one more info: my smartmeter is connected via lte insteaf of powerline to vkw and i have enabled the 1/4 hour transmission in vorarlbergnetz portal.

Exolor, 2022-04-17:
I also had these issues, but i can´t code and so i added some filters to remove the bad values. 
For live watt usage i just cut it off at 5kW as this is the most i use, and for the energy dashboard i use a median 

`    filters:
    - median:
        window_size: 15
        send_every: 5
        send_first_at: 4`
        
Works good enough for me, but just a workaround.

cococheaf, 2022-04-18:
Thank you Exolor, will try that.
Will report back if that workaround also works for me.

Exolor, 2022-04-18:
```
 - platform: template
    id: meter01_active_power_plus
    name: meter01_active_power_plus
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: "power"
    state_class: "measurement"
    filters:
     - lambda: |-
            float MIN_VALUE = 0.0;
            float MAX_VALUE = 5000.0;
            if (MIN_VALUE <= x && x <= MAX_VALUE) return x;
            else return {};

  - platform: template
    id: meter01_active_energy_plus
    name: meter01_active_energy_plus
    unit_of_measurement: Wh
    accuracy_decimals: 0
    device_class: "energy"
    state_class: "total_increasing"
    filters:
    - median:
        window_size: 15
        send_every: 5
        send_first_at: 4
```

This would be what my sensors look like, i still get a wrong value with the power sensors sometimes, or i forgot i used the oven at 4 in the morning. 
But i dont really care about the live power usage as long as the graph is readable for the rest. 

If you still get incorrect data with the energy sensors, you can increase the window_size. As they dont change a lot, it doesnt really matter anyways.

deviant-aut, 2022-09-21:
One wrong bit in the payload can mess up all values.

I would suggest to check the encrypted  APDU.  We could verify the APDU by checking a Datatype.
eg. we know the HEX value and the postition of the octet datatype. If it isn't correct we can just drop the APDU.

At least that should work for [EVN](https://www.netz-noe.at/Download-(1)/Smart-Meter/218_9_SmartMeter_Kundenschnittstelle_lektoriert_14.aspx). I gues it is the same for other providers.

DomiStyle, 2022-09-22:
Since actual spec sheets for the protocol(s) are available now and there is much less guessing going on I rewrote the entire parsing logic so these kind of errors should hopefully be detected by the M-Bus checksum and the length fields.

The changes are in the master branch if anybody wants to give it a try.

cococheaf, 2022-10-07:
Nope - exchanged all the files, recreated the image and uploaded to the esp. same problem as before.

cococheaf, 2022-10-07:
> @cococheaf [1] The smart meter is probably really providing invalid
> data then. It's unlikely that errors are not detected by the checksum
> multiple times.

No problem.

I keep the meridian function running, that works very well.

Thank you - Rgds - Andre

Links:
------
[1] https://github.com/cococheaf
[2] 
https://github.com/DomiStyle/esphome-dlms-meter/issues/16#issuecomment-1271892580
[3] 
https://github.com/

Exolor, 2022-10-09:
> Nope - exchanged all the files, recreated the image and uploaded to the esp. same problem as before.

Same for me, didnt seem to change anything, so as Domi said, prob a problem with the smart meter.

deviant-aut, 2022-10-09:
> > Nope - exchanged all the files, recreated the image and uploaded to the esp. same problem as before.
> 
> Same for me, didnt seem to change anything, so as Domi said, prob a problem with the smart meter.

No it is not. We change hardware to a different esp and problems are gone. 
Problem only occurs with esp32 dev board from az delivery

Exolor, 2022-10-09:
> 

Could very well be, i dont use one from AZ but another chinese board. 
Which esp are you using now?

Exolor, 2022-10-09:
> No it is not. We change hardware to a different esp and problems are gone. Problem only occurs with esp32 dev board from az delivery

> No it is not. We change hardware to a different esp and problems are gone. Problem only occurs with esp32 dev board from az delivery

Could very well be, i dont use one from AZ but another chinese board.
Which esp are you using now?

deviant-aut, 2022-10-09:
I can confirm running on ESP32-CAM and ESP32 D1 Mini (random manufacturer from amazon).
connected rx,tx,3.3v and gnd to a mikroe m-bus clicker slave.

The ESP32 node mcu from AZ was only able to receive data when 3.3v was not connected but still unstable.

I don't realy understand how this can happen as the code should log an error and drop the invalid payload.
As the error did not occure at my home i did no further investigation.

cococheaf, 2023-03-02:
I think i have found the problem regarding the abnormal data.
The esp32-dev module is now running with no issues, i have changed the RX Pin from GPIO4 to GPIO35.
That fixed the data spikes, i had no spike since one month now.

Maybe someone else with an esp32-dev module can verify that?

 - Andre

vagabond235, 2023-04-14:
Just to confirm a working set-up (at least for the last ~24h :)) using the following configuration:
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 2400
  rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  id: mbus

This works just fine, thanks for the hint! No more spikes, no more errors or warnings in the webinterface.

Hardware used: AzDelivery ESP32 D1 Mini NodeMCU WiFi Modul + Bluetooth(ESP32-wroom-32) with a MIKROE : M-BUS SLAVE CLICK connected to a KAIFA MA309M (Tinetz).
Flo
[image]

I also tried to change tx to GPIO35 - and got the following error message:
uart: [source config/meter01.example.yaml:37]
  
    GPIO35 (34-39) does not support output pin mode.
rx would probably work on GPIO35.

zibous, 2023-05-07:
@cococheaf 

Also located in Höchst.
Where did you buy the MBus level converter?

unsign, 2023-05-07:
@zibous I'm located in Luschnou ;) .. Have a m-bus slave click from mikroe left over, if you need one :)

zibous, 2023-05-07:
Hi

> @zibous I'm located in Luschnou ;) .. Have a m-bus slave click from mikroe left over, if you need one :)

Super, würde den nehmen - wie kann ich den bekommen ?
Projekte/Kontakt siehe https://github.com/zibous

unsign, 2023-05-07:
> Hi
> 
> 
> 
> > @zibous I'm located in Luschnou ;) .. Have a m-bus slave click from mikroe left over, if you need one :)
> 
> 
> 
> Super, würde den nehmen - wie kann ich den bekommen ?
> 
> Projekte/Kontakt siehe https://github.com/zibous
> 
> 
> 
> 
> 
> 

u got mail ... ruf einfach kurz an ✌🏽

cococheaf, 2023-05-07:
Sorry - hab erst vorher die Mails durchgeschaut - hat sich aber in dem Fall schon erledigt :)
Viel Spaß beim Basteln ;)

LG