DomiStyle/esphome-dlms-meter #17: Payload length is too big for received data

smiyc, 2022-08-05:
Hi,

I have a ESP32 devboard and an T210-D from EVN.
I'm facing this error: 

[espdm:053] | Payload length is too big for received data

This is my yaml:

esphome:
  name: meter01
  platform: ESP32
  board: esp32dev
  includes:
    - ../esphome-dlms-meter

wifi:
  ssid: "XX"
  password: "XXXXXXXX"

# Enable logging
logger:
  level: VERBOSE
  tx_buffer_size: 2048

# Enable Home Assistant API if not using MQTT
api:
  encryption:
    key: "ujZvNH2U9kMg/o/rFvNO2NvVaXeFQJXp0XXXXXXXXX"

ota:

web_server:
  port: 80

uart:
  tx_pin: GPIO4
  rx_pin: GPIO36
  baud_rate: 2400
  parity: EVEN
  data_bits: 8
  stop_bits: 1
  rx_buffer_size: 2048
  id: mbus

sensor:
  - platform: template
    id: meter01_voltage_l1
    name: meter01_voltage_l1
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: "voltage"
    state_class: "measurement"
  - platform: template
    id: meter01_voltage_l2
    name: meter01_voltage_l2
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: "voltage"
    state_class: "measurement"
  - platform: template
    id: meter01_voltage_l3
    name: meter01_voltage_l3
    unit_of_measurement: V
    accuracy_decimals: 1
    device_class: "voltage"
    state_class: "measurement"

  - platform: template
    id: meter01_current_l1
    name: meter01_current_l1
    unit_of_measurement: A
    accuracy_decimals: 2
    device_class: "current"
    state_class: "measurement"
  - platform: template
    id: meter01_current_l2
    name: meter01_current_l2
    unit_of_measurement: A
    accuracy_decimals: 2
    device_class: "current"
    state_class: "measurement"
  - platform: template
    id: meter01_current_l3
    name: meter01_current_l3
    unit_of_measurement: A
    accuracy_decimals: 2
    device_class: "current"
    state_class: "measurement"

  - platform: template
    id: meter01_active_power_plus
    name: meter01_active_power_plus
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: "power"
    state_class: "measurement"
  - platform: template
    id: meter01_active_power_minus
    name: meter01_active_power_minus
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: "power"
    state_class: "measurement"

  - platform: template
    id: meter01_active_energy_plus
    name: meter01_active_energy_plus
    unit_of_measurement: Wh
    accuracy_decimals: 0
    device_class: "energy"
    state_class: "total_increasing"
  - platform: template
    id: meter01_active_energy_minus
    name: meter01_active_energy_minus
    unit_of_measurement: Wh
    accuracy_decimals: 0
    device_class: "energy"
    state_class: "total_increasing"

  - platform: template
    id: meter01_reactive_energy_plus
    name: meter01_reactive_energy_plus
    unit_of_measurement: Wh
    accuracy_decimals: 0
    device_class: "energy"
    state_class: "total_increasing"
  - platform: template
    id: meter01_reactive_energy_minus
    name: meter01_reactive_energy_minus
    unit_of_measurement: Wh
    accuracy_decimals: 0
    device_class: "energy"
    state_class: "total_increasing"

text_sensor:
  - platform: template
    id: meter01_timestamp
    name: meter01_timestamp

mqtt:
  broker: 10.0.1.12
  id: mqtt_broker
  username: "mqtt"
  password: "gvdScfLM5DbQ7KBRXXXXXXXXX"

custom_component:
  - lambda: |-
      auto dlms_meter = new esphome::espdm::DlmsMeter(id(mbus));
      byte key[] = {0x97, 0x64, 0x0D, 0x7B, 0x3A, 0x21, 0x9D, 0xB0, 0xE4, 0xB5, 0x08, 0xXX, 0xXX, 0xXX, 0xXX, 0xXX};
      dlms_meter->set_key(key, 16);
      dlms_meter->set_voltage_sensors(id(meter01_voltage_l1), id(meter01_voltage_l2), id(meter01_voltage_l3));
      dlms_meter->set_current_sensors(id(meter01_current_l1), id(meter01_current_l2), id(meter01_current_l3));
      dlms_meter->set_active_power_sensors(id(meter01_active_power_plus), id(meter01_active_power_minus));
      dlms_meter->set_active_energy_sensors(id(meter01_active_energy_plus), id(meter01_active_energy_minus));
      dlms_meter->set_reactive_energy_sensors(id(meter01_reactive_energy_plus), id(meter01_reactive_energy_minus));
      dlms_meter->set_timestamp_sensor(id(meter01_timestamp));
      dlms_meter->enable_mqtt(id(mqtt_broker), "meter01/data");
      return {dlms_meter};

When I manually decrypted the packet

6801016853FF000167DB085341475905E746DC81F82000191D23079837D1DDD93A50ACC5956E84BDFFA775EAB83F27A42E334FCD4C17AE131428DF6B716BCB1E5B7BDACE4A56FEF39136A51AB1A2F7A8D0CF8CF12BB0BD2826D7A2474B7703A25F5F023684732BAB2E34FDD9E81C1A0558D4699D71EAF549714A095E3A8FB211C37056EC076D249737A4BCD9B99F1A41C21A3604B3101783C4EBDC3CC42F947DD47DB83B6EDE96D7AA19EFA21BCC3248E583A5F9AC47CA77D27BA7CAC05421FDC5AA9D5FAF5C11D681A295CBBEA3FC91D6299F5002F016FE32BC954AF5718C37DA2FCB540B5AD3A48AC24B76221DE3FBD5FA6B0EB7864778C6FD5ADA555111375CA85B644F6916680D0D6853FF110167A1F6ECD20D9E4995A916

with https://github.com/ric-geek/DLMS-APDU-Encrypter-Decrypter I get:

APDU: 0f806f77230c07e608050511132800ff88820223090c07e608050511132800ff888209060100010800ff06010bd2a202020f00161e09060100020800ff060000004802020f00161e09060100010700ff06000004ef02020f00161b09060100020700ff060000000002020f00161b09060100200700ff12090c02020fff162309060100340700ff1208f302020fff162309060100480700ff12090602020fff1623090601001f0700ff12003702020ffe162109060100330700ff1200df02020ffe162109060100470700ff12015302020ffe1621090601000d0700ff1003e402020ffd16ff090c31373832f9d0b4eb33c6295c875841717b334fb322a385
TAG: cc4168a058ea98312aa08c22

[image]

smiyc, 2022-08-28:
according to the EVN PDF https://www.netz-noe.at/Download-(1)/Smart-Meter/218_9_SmartMeter_Kundenschnittstelle_lektoriert_14.aspx
the mubs packet should start with 68FAFA68.
Mine starts with 68010168, so I think the problem is somehow related to the m-bus slave click board.

Has anybody a working config for EVN and T210-D?
I have searched related issues and set parity: EVEN and remove R5 from the m-bus- slave click board as seen in #7  , without success.

...in the mean time, "Österreichs Energie" ordered Ginzinger to develop an "einheitlichen Schnittstelle für Smart Meter" --> https://www.ginzinger.com/de/wissen-events/techtalk/ginzinger-entwickelt-smart-meter-schnittstelle/
which could be available this autumn. 

Antworten gerne auch auch auf deutsch

deviant-aut, 2022-09-24:
@smiyc  I have a fork of an older version of this repo running with T210-D. 
[I just uploaded my fork](https://github.com/deviant-aut/esphome-dlms-meter?organization=deviant-aut&organization=deviant-aut)
If I find time I want to add the latest changes for fail safe parsing. Still, for me it's running rock solid since several months.

#12 someone mentions the wrong value in length field is an issue with the T210-D from NETZ-NOE so I just set an static length value.

ra-mi, 2023-01-26:
@deviant-aut - your esphome-dlms-meter ist the only which give me some data - thx! , but after the "Current_L2" i get an:
[espdm:125]:  15
[espdm:126]:  12
[espdm:127]:  7
[espdm:128]:  231
[espdm:129]:  9
[espdm:396]: Unsupported OBIS data type

any ideas? T210-D(Noe)

smiyc, 2023-11-24:
Using a fronius smartmeter now...

closed 2023-11-24 by smiyc