DomiStyle/esphome-dlms-meter #19: Esphome 12.2022 & HA - MBUS: Start bytes do not match & Frame too big for received data

dr-acula-flo, 2022-12-14:
Hey all,
the code and hardware was working great for me the last few months (thanks for that btw!) but the recent 12.2022 Esphome upgrade, which I run as a Home Assistant add-on, seems to have broken something. I have the Kaifa MA110M smart meter (IKB).

As others, I'm using a m-bus-slave-click for the connection to the smart meter. I have a DFRobot FireBeetle 2 ESP32-E IoT Microcontroller board (this one: https://www.dfrobot.com/product-2231.html) because of it's low energy consumption in deep sleep as all parts are powered off a battery.

So, I updated the Esphome add-on to 12.2022 today and also pushed the available update to the devices. After those updates were done, I now get the following two error messages in the Esphome log of my board:

[15:05:16][E][espdm:048]: MBUS: Start bytes do not match
[15:05:16][E][espdm:064]: MBUS: Frame too big for received data

My uart configuration is:
`uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 2400
  rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  id: mbus`

I've tried adjusting the `rx_buffer_size` to 2048 and also tried other things that were mentioned in other issues (i.e. adding parity, data and stop bits) but that didn't do the trick.

So, does anybody have an idea what might have changed or how I go about debugging it?
Thanks in advance, and I hope I've provided enough information.

DomiStyle, 2022-12-14:
I can reproduce this on 2022.12 but I'm gonna have to check what changed.

For now, 2022.11.5 works fine.

theexit, 2022-12-19:
I had also an issue since update to 12.2022.
[13:23:12][E][espdm:038]: Received packet with invalid size
[13:23:12][E][espdm:038]: Received packet with invalid size
[13:23:13][E][espdm:038]: Received packet with invalid size
Hope this helps a little bit.

theexit, 2022-12-22:
#21 added fix... is working for me!

dr-acula-flo, 2022-12-22:
> #21 added fix... is working for me!

Thanks a lot to the both of you. It wouldn't work for me (IKB) with just adding the delay. While the "Frame too big for received data" didn't show up in the log anymore, the "Start bytes do not match" was still there. I had to change my uart config from:
```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 2400
  rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  id: mbus
```
to
```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 2400
  #rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
  parity: EVEN
  data_bits: 8
  stop_bits: 1
  rx_buffer_size: 2048
  id: mbus
```
I haven't checked what actually made the difference, but the above-mentioned fix and this config change worked in my case.

theexit, 2022-12-23:
> > #21 added fix... is working for me!
> 
> Thanks a lot to the both of you. It wouldn't work for me (IKB) with just adding the delay. While the "Frame too big for received data" didn't show up in the log anymore, the "Start bytes do not match" was still there. I had to change my uart config from:
> 
> ```
> uart:
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   baud_rate: 2400
>   rx_buffer_size: 1024 # Needed to receive the large packets send by the smart meter
>   id: mbus
> ```
> 
> to
> 
> ```
> uart:
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   baud_rate: 2400
>   parity: EVEN
>   data_bits: 8
>   stop_bits: 1
>   rx_buffer_size: 2048
>   id: mbus
> ```
> 
> I haven't checked what actually made the difference, but the above-mentioned fix and this config change worked in my case.

That's true. You must change the rx_buffer_size to 2048.
But you can leave data_bits and stop_bits because these are the default values.
I also have no parity settings.

dr-acula-flo, 2023-01-29:
> That's true. You must change the rx_buffer_size to 2048. But you can leave data_bits and stop_bits because these are the default values. I also have no parity settings.

Thanks! I've tried removing them one by one but it wouldn't work without the `parity: EVEN` setting enabled, at least in my case. It would still error out with `Start bytes do not match`.

StefanTT, 2025-06-28:
I also had to specify "parity: EVEN" for the Kaifa MA309M (from Salzburg.ag)

schoerg, 2025-09-21:
for which kaifa device? MA309MH4LAT1?

StefanTT, 2025-09-22:
Yes, the MA309MH4LAT1