DomiStyle/esphome-dlms-meter #27: hello could you please explain how too install it on esph home on homeassitant

rogeruncle12, 2023-04-06:


DomiStyle, 2023-04-07:
There's an easy to follow guide on how to setup ESPHome here: https://esphome.io/guides/getting_started_hassio.html

In case you need to setup Home Assistant as well, it's easiest to do it on a Raspberry Pi or similar and use their provided images: https://www.home-assistant.io/installation/

rogeruncle12, 2023-04-08:
hello thank you for your reply

i am  familar with this site, but i dont know what did you men with 
"Clone this repository into your ESPHome config files"

Exolor, 2023-04-15:
> hello thank you for your reply
> 
> i am familar with this site, but i dont know what did you men with "Clone this repository into your ESPHome config files"

Download the code as zip, unpack it and put the files into config/esphome/esphome-dlms-meter