DomiStyle/esphome-dlms-meter #34: Decryption key format

valki2, 2023-10-17:
Hi!
I could need a little hint.... it looks like I get some data via  Mbus but cannot decode correctly. I guesss its a user-error :)

```
[20:52:41][W][component:204]: Component <unknown> took a long time for an operation (0.07 s).
[20:52:41][W][component:205]: Components should block for at most 20-30ms.
[20:52:42][VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=104866 (now=164866)
[20:52:46][W][component:204]: Component <unknown> took a long time for an operation (3.00 s).
[20:52:46][W][component:205]: Components should block for at most 20-30ms.
[20:52:46][VV][scheduler:226]: Running interval 'update' with interval=60000 last_execution=106814 (now=168623)
[20:52:46][V][espdm:526]: 68.FA.FA.68.53.FF.00.01.67.DB.08.4B.46.4D.10.10.02.45.50.82.01.09.21.00.00.09.00.13.30.5A.F8.21.5C.96.21.FE.1F.E1.C3.D1.8D.61.98.7D.8C.75.93.4F.BC.C4.B5.59.D8.91.B8.86.FF.45.27.39.BE.B5.7F.D0.19.E6.81.B7.38.30.D4.2F.67.EA.B8.06.0F.DB.6B.B0.38.8A.18.E3.6D.CC.AB.58.E1.5E.1A.F2.67.A4.7D.48.B1.07.E3.F2.4D.57.63.1E.E1.5D.CB.8E.B1.57.4C.7B.72.12.59.9D.37.39.A6.F7.EA.F0.AF.B4.3A.55.F6.52.30.3A.EA.E2.0F.9B.63.91.F0.E4.03.77.B1.D5.FB.F8.65.80.72.06.CC.4E.95.91.69.38.B1.46.3B.C6.89.D8.21.8B.93
[20:52:46][V][espdm:041]: Parsing M-Bus frames
[20:52:46][V][espdm:048]: MBUS: Parsing frame
[20:52:46][V][espdm:048]: MBUS: Parsing frame
[20:52:46][V][espdm:091]: Parsing DLMS header
[20:52:46][V][espdm:118]: DLMS: Message length > 127
[20:52:46][V][espdm:146]: Decrypting payload
[20:52:46][V][espdm:167]: Check: ESP32
[20:52:46][E][espdm:182]: OBIS: Packet was decrypted but data is invalid
```
From the providers website, i got my key for the Kaifa MA110M which has 32 characters. 
I converted each character:
1 > 0x01
2 > 0x02
...
A > 0x0A as well as 0x10
etc.
Therefore, I increased the arrays lenght: dlms_meter->set_key(key, 32); 

Both variants didnt work, so I tried using a 16 length array and converting double characters... no change.

To rule out that I bark up the wrong tree - what is the correct way to config that decryption key?

valki2, 2023-10-18:
ok looks like the data is truncated as there is no "16" (Mbus Stop) at the end.

ert666, 2023-12-15:
@valki2 
So your code would be...
...
      uint8_t key[] = {0xD6, 0x58, 0x34, 0x5D, 0xF6, 0x20, 0xA2, 0x83, 0x9E, 0x5A, 0xD6, 0xA6, 0xD1, 0xB8, 0x51, 0x35};
      dlms_meter->set_key(key, 32; // Pass your decryption key and key length here
...
?

kenci, 2024-01-07:
@ert666 no, it should be `dlms_meter->set_key(key, 16);` because there are 16 bytes.