DomiStyle/esphome-dlms-meter #39: Power ESP via M-Bus

luusl, 2024-04-29:
Has anyone ever tried to supply the ESP32 with power via the M-Bus? I don't have a socket in my switch cabinet and don't want to install any cables now. According to the specification, there should always be 24-36V DC on the M-Bus, so it should be possible with a Zener diode (I have some BZX55C9V1 lying around here) and a step-down converter (LM2596 available).

SimonFischer04, 2024-04-29:
Yes, but according to spec also only 6ma. So quite hard to get it working.
At least consumption with wifi (even more during sending of data)  seems not really possible in my opinion.

luusl, 2024-04-30:
True, I didn't think of that. So, laying cables after all. Thank you.

closed 2024-04-30 by luusl

Soundrico, 2024-04-30:
Hey there, I guess you can power an esp32 via M-Bus.
As I remember there is somewhere in the m-bus specs written that each of multiple bus-members limited to 6mA. ...In a total of 48mA or something like that. So the meter internal m-bus power supply is able to deliver more then 6mA.With a bare Esp32 and the m bus chip it should work- there exists a YouTube video that shows such a constellation. I guess it was a dlnr esp home config - I can't remember exactly- google for it
BTW: this project works fine for me in my smarthome since over half a year - congratulations to the developer

SimonFischer04, 2024-04-30:
> True, I didn't think of that. So, laying cables after all. Thank you.

Yeah, currently trying around with this also. As it would save quite a lot of hassle here...
But unfortunately (at least with wifi) i can't think of how it could  be sufficient.

SimonFischer04, 2024-04-30:
> As I remember there is somewhere in the m-bus specs written that each of multiple bus-members limited to 6mA. ...In a total of 48mA

Please provide link if you can as I can't find something like this.

PS: also (at least with my smartmeter) wrong, as I tested it and the smartmeter hard-cuts off power if drawing >~10ma

> there exists a YouTube video that shows such a constellation

Please link, cant find. Dsmr vs dlms should not matter. 
Also not all "m-bus" are equal. Some labeled as "P1" port have a constant 5v power line on the same connector with easily enough current.