DomiStyle/esphome-dlms-meter #40: Start bytes do not match nach Monaten ohne Probleme

DaPeace, 2024-05-06:
Hi, gestern hab ich feststellen müssen, das mein Zähler keine Daten mehr liefert. 
Ich hab dann ein Upgrade gemacht von ESPHome und entsprechend auch den Code auf dem ESP aktualisiert aber er bringt immer noch "06:14:24 [E]	[espdm:053]	MBUS: Start bytes do not match". Er bekommt alle 5 Sekunden Daten, kann sie aber nicht entschlüsseln.
Das ganze lief jetzt für einige Monate völlig problemlos. Hat sich irgendetwas geändert bei der Tinetz? Geht es noch mehr so wie mir?
Grüsse, Jan

boskape, 2024-08-11:
Hi

Ich habe das selbe Problem
Habe einen ESP32 welche ich vor 2 Jahren noch unter IO Broker konfiguriert habe.
Da konnte ich heute noch auf der Website des ESP die Smartmeter Daten auslesen

Leider hatte ich die idee auf HA mit aktuellem ESPHome zu wechseln
Habe den ESP eingebunden, die yaml konfiguriert 
Doch seitem sehe ich nur mehr Start byte doesn match
auch den Buffer auf 2048 hoch gedreht
ohne erfolg

Hat jemand eine Idee
Hat sich an der ESPHome Firmware etwas geändert
Komme ich irgendwie wieder auf die alte esphome version ?

boskape, 2024-08-11:
Hello
habe auf esphome 2022.11 zurück geflasht
und siehe da
es läuft wieder
[image]

boskape, 2024-08-20:
> Hi,
> sorry falls das nach (fremd) Werbung klingt, aber:
> Ich arbeite gerade an einem Update der Komponente und hoffentlich später mal Integration in esphome nativ:
> SimonFischer04/esphome#1<https://github.com/SimonFischer04/esphome/issues/1>
>
> -

Danke für de mich freuen
Eventuell kann man es gelich erweitern um auch den Gaszähler per reed contakt mit abzugreifen

Ich denke bei vielen ist gas und Stromzähler nahe bei einander

Lg
Peter

SimonFischer04, 2024-08-20:
> Eventuell kann man es gelich erweitern um auch den Gaszähler per reed contakt mit abzugreifen

Um überhaupt sagen zu können ob das möglich ist / wieviel Aufwand wären schon ein paar mehr Infos nett :)

Ich selber hab keinen Gaszähler.

Nur weil der nebenbei hängt heist das nicht das man das "einfach erweitern" kann.

Um welchen Gaszähler handelt es sich?
Gibt es dazu irgendwo eine Schnittstellenbeschreibung.

Falls der ein anderes Protokoll verwendet, macht es keinen Sinn das hier mit in der Komponente einzubauen.

boskape, 2024-08-21:
HAllo

Nun da  kann ich selbst nur auf beispiele verweisen da ich mich mit der Programmierung dieser esp32 und esp8266 nicht so auskenne

https://smarthome.ms1.de/projekte/gasverbrauch/

vielleicht sind strom und gaszähler nicht immer nebeneinander und dann würde man 2 stück esp benötigen

aber wenn es immer die selbe software wäre würde dies es auch einfacher machen

lg

SimonFischer04, 2024-08-21:
> die selbe software wäre

Wie bereits beschrieben. Einfach ausgedrückt: das eine ist ein **STROM**- das andere **GAS**- Zähler (mal abgesehen davon das es wiederum unterschiedliche Stromzähler Modelle gibt). Aka die gleiche Software wird nicht unterschiedliche Dinge lesen.

> Um welchen Gaszähler handelt es sich?

Wenn ich schon das googlen übernehmen muss, wäre eine Antwort auf die Frage sehr nett / hilfreich.
Ansonsten wirds schwierig hier weiterzukommem. Auch Gaszähler sind unterschiedlich.

boskape, 2024-08-23:
Sorry für die Verzögerung
Es handelt sich um einen Elster Instroment
BK-G4

SimonFischer04, 2024-10-11:
Ok könnte funktionieren.
Kannst es ja mal mit der Anleitung aus deiner vorherigen Nachricht probieren.