DomiStyle/esphome-dlms-meter #41: Possible to Read Water-Meters?

MasterOfPuppets111, 2024-05-20:
Hello, I have a Watermeter mit a MBus (wired) Interface, possible to use this?

Best regards Pierre

MasterOfPuppets111, 2024-08-20:
Hello Simon,

Thank you for your answer.

It is a Sensus Mbus HRI-B Module.
Here is the documentation, I hope this helps:
https://cdn.shopify.com/s/files/1/0435/3399/2089/files/DE_EN_ES_FR_-_HRI_Manual.pdf?v=1705053742

Best regards Pierre

SimonFischer04, 2024-08-20:
Hello,

I dont fully undertand this documentation.

But it seems to use a completely different protocoll to the power meters (no dlms encryption, ...) So does not make sense in this component.

You can try to connect an mbus slave to uart board like with the power meter.
And the use esphome uart debug setting to see  what data you actually get from the device:

https://www.esphome.io/components/uart.html#uart-debugging