DomiStyle/esphome-dlms-meter #43: Help with M-Bus Slave

erik-toth, 2024-07-17:
Hello!
I used a diffrent M-Bus Slave board and wanted to know if anyone has any experience with the "[replica board](https://a.aliexpress.com/_ExTx9lN)". 

[image]

I have set up everything like the Hardware Installation in the read me recommends, but as a value in HA i just get "unknown". I already let it run for 1 hour, just to be sure. Now me question is: Might it be the board or something different I've done wrong.

Best regards Erik!

SimonFischer04, 2024-08-19:
Hi,
I'm working on updating the component and eventual native integration into esphome.
You could try my version and see if it helps: https://github.com/SimonFischer04/esphome/issues/1

SimonFischer04, 2024-08-20:
So as you got it working, this issue can be closed now?

closed 2024-08-21 by erik-toth