DomiStyle/esphome-dlms-meter #47: Bug since esphome 2025-02: "custom" component has been removed

pos-ei-don, 2025-02-19:
```
Failed config

custom_component: [source /config/esphome/smartmeter-esp.yaml:307]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  - lambda: |-
      
      auto dlms_meter = new esphome::espdm::DlmsMeter(id(mbus));
      
      uint8_t key[] = {};
      dlms_meter->set_key(key, 16); // Pass your decryption key and key length here
      
      dlms_meter->set_voltage_sensors(id(meter01_voltage_l1), id(meter01_voltage_l2), id(meter01_voltage_l3)); // Set sensors to use for voltage (optional)
      
      dlms_meter->set_current_sensors(id(meter01_current_l1), id(meter01_current_l2), id(meter01_current_l3)); // Set sensors to use for current (optional)
      
```

Can anyone update the tescription to use it as external component?

cococheaf, 2025-02-20:
The problem is that the ESP Home custom_component was replaced by the external_component.

Here is a link to the announcement of the removal:
[](https://esphome.io/guides/contributing#a-note-about-custom-components)

And here is the link to the external component documentation:
[](https://esphome.io/components/external_components)

@DomiStyle - can you please check the code and rewrite?
Thank you - greeting aus Höchst ;)

cococheaf, 2025-02-20:
I am testing it right now, as for now it seems to work with any other smartmeter with dlms encryption. Will update in a few minutes

ximex, 2025-02-20:
sry hat the wrong pr id before. the 8009 is correct

pos-ei-don, 2025-02-20:
Hm, the code "external_components" does not work here,and i cannot find the error?!?

```
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "/config/esphome/smartmeter-v3.yaml", line 14, column 3
expected <block end>, but found '?'
  in "/config/esphome/smartmeter-v3.yaml", line 30, column 3
```

Any idea?

cococheaf, 2025-02-20:
So i got it perfectly working again with this yaml configuration.
Running on a ESP32DEV board with MBus Slave Click and Kaifa MA309M:

```yaml

esphome:
  name: smartmeter
  comment: "'NodeMCU ESP32 Dev' mit MikroE-MBUS-SlaveClick an Kaifa MA309M Stromzähler"
  friendly_name: Stromzähler

esp32:
  board: esp32dev

api:

ota:
 - platform: esphome
   password: "yourotasecret"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: w.x.y.z
    gateway: w.x.y.z
    subnet: w.x.y.z
    dns1: w.x.y.z
  ap:
    ssid: "yourfallbackssid"
    password: "yourwifipwd!"

mqtt:
  broker: w.x.y.z
  username: yourmqttuser
  password: yourmqttpwd
  id: mqtt_broker

captive_portal:

logger:
  # level: VERBOSE
  # tx_buffer_size: 1024

web_server:
  port: 80
  version: 2
  local: true

external_components:
  - source: github://SimonFischer04/esphome@dlms-meter
    components: [ dlms_meter ]

uart:
  rx_pin: GPIO33
  tx_pin: GPIO32
  baud_rate: 2400
  data_bits: 8
  stop_bits: 1
  rx_buffer_size: 2048
  id: mbus

dlms_meter:
  decryption_key: "yourencryptionkey"

sensor:
  - platform: dlms_meter
    voltage_l1:
      name: "smartmeter_voltage_l1"
    voltage_l2:
      name: "smartmeter_voltage_l2"
    voltage_l3:
      name: "smartmeter_voltage_l3"
    current_l1:
      name: "smartmeter_current_l1"
    current_l2:
      name: "smartmeter_current_l2"
    current_l3:
      name: "smartmeter_current_l3"
    active_power_plus:
      name: "smartmeter_active_power_plus"
    active_power_minus:
      name: "smartmeter_active_power_minus"
    active_energy_plus:
      name: "smartmeter_active_energy_plus"
    active_energy_minus:
      name: "smartmeter_active_energy_minus"
    reactive_energy_plus:
      name: "smartmeter_reactive_energy_plus"
    reactive_energy_minus:
      name: "smartmeter_reactive_energy_minus"

text_sensor:
  - platform: dlms_meter
    timestamp:
      name: "smartmeter_timestamp"

```

mariopenzendorfer, 2025-02-20:
I adapted the existing code to use external components, together with another fork of this project to get it compatible for my SmartMeter (as @pos-ei-don mentioned).

Was planning on re-intergrating the "old" code to fully support all sorts of devices again but did not have the time yet.

I will keep you updated on my progress.

cococheaf, 2025-02-20:
> Just saw that a fork of this project did this, unfortunately they also changed the code to support only a certain type of smartmeter. So I cannot use this code.
> 
> Here is a Reference to the other code, maybe it helps to update this code? Or maybe [@mariopenzendorfer](https://github.com/mariopenzendorfer) can provide a PR for this? [mariopenzendorfer@36bee98](https://github.com/mariopenzendorfer/esphome-smartmeter/commit/36bee98c26d75a35417d0856a789496d9c340935)

There are two configs on Simon´s site - one for his particular meter and one for all others.

pos-ei-don, 2025-02-20:
> So i got it perfectly working again with this yaml configuration. Running on a ESP32DEV board with MBus Slave Click and Kaifa MA309M:

This code works perfectly here aswell!!

ximex, 2025-02-20:
@mariopenzendorfer would be nice if we could put your changes you need also into the PR from @SimonFischer04 instead of having different implementations

mariopenzendorfer, 2025-02-20:
@ximex I agree, will have a look on Simon's project (it might be working for me too after all).

Also, he implemented the ESPHome component structure in a different way than I did, which looks quite interesting.

I don't know how actively this repo is still maintained, but getting all the implementations merged into one thats working and easily configurable with the new external_components structure definitly makes sense.

ximex, 2025-02-20:
@cococheaf is there a reason for `rx_buffer_size: 2048` as @SimonFischer04 has documented `1024`

pos-ei-don, 2025-02-20:
In problem still remains:
With the new structure my
combined sensor (plus and minus) does not work anymore.
I get 
> Couldn't find ID 'active_power_plus'. Please check...

The code which i used in the old version and worked is:

```
  - platform: template
    id: smartmeter_active_power
    name: meter01_active_power
    unit_of_measurement: W
    accuracy_decimals: 0
    device_class: "power"
    state_class: "measurement"
    update_interval: 1s
    lambda: |-
        return (id(active_power_plus).state - id(active_power_minus).state);

```

Can you help me to find a solution to work with the numbers ?

cococheaf, 2025-02-20:
> [@cococheaf](https://github.com/cococheaf) is there a reason for `rx_buffer_size: 2048` as [@SimonFischer04](https://github.com/SimonFischer04) has documented `1024`

@ximex - no - i just had problems with spikes. I have then changed gpio ports and buffer sizes and the errors went away.
I for myself will keep that setting as it is working wonderfully since 2 years.

cococheaf, 2025-02-20:
> In problem still remains: With the new structure my combined sensor (plus and minus) does not work anymore. I get
>
>
> The code which i used in the old version and worked is:
>
> ```
>   - platform: [...]

i would create a helper in home assistant itself. makes more sense imho.

pos-ei-don, 2025-02-20:
Sorry, ignore my last post. found the copy/paste error

pos-ei-don, 2025-02-20:
[cococheaf](https://github.com/cococheaf)
> i would create a helper in home assistant itself. makes more sense imho.

Thank you, but no! I write this value via MQTT directly to my inverter! 
Works great, even id home assistant is not available (rebooting, ...)

This makes this solution so much better than any other bought smartmeter-reader.

I just need OBIS code 16.7.0 aswell ( ;-) )
https://github.com/DomiStyle/esphome-dlms-meter/issues/46

SimonFischer04, 2025-02-20:
> @ximex I agree, will have a look on Simon's project (it might be working for me too after all).
>
> Also, he implemented the ESPHome component structure in a different way than I did, which looks quite [...]

@mariopenzendorfer 

Which provider do you have?

Yes. Heavily inspired by the dsmr component. Hope this makes implementing more sensors easier / less copy paste.

Which repo do you mean? Domis or mine?
For me: I work on this in my free time, so obviously not everything will happen instantly but if issues arise i will try to fix them.

Agreed, this is why i already merged a few implementations into mine. And now trying to get this upstreamed into esphome itself.

SimonFischer04, 2025-02-20:
> [cococheaf](https://github.com/cococheaf)
>
> Thank you, but no! I write this value via MQTT directly to my inverter!
> Works great, even id home assistant is not available (rebooting, ...)
>
> This makes [...]

@pos-ei-don 

Where did you get this obis code from?

Have you checked that your smartmeter actually sends it? I have not yet seen any that sends this one.

SimonFischer04, 2025-02-20:
For all you successfully testing:

Would appreciate a review, approve, upvote of the esphome prs, so it gets more traction there (without any annoying "when merge" comments...)

:)

pos-ei-don, 2025-02-20:
[@SimonFischer04]
> Have you checked that your smartmeter actually sends it? I have not yet seen any that sends this one.

Good question: how can i check this?
I had an comercially produced device which gave mit this OBIS code. So i think it should be there.
But it can be a caclulated value aswell...

ximex, 2025-02-20:
@pos-ei-don do you have a technical docu about your product?

pos-ei-don, 2025-02-20:
[ximex](https://github.com/ximex): I am a bit confused,
This does not work for me, so i tested the code from
https://github.com/DomiStyle/esphome-dlms-meter/issues/47#issuecomment-2671025166

> maybe this PR helps someone: [esphome/esphome#8009](https://github.com/esphome/esphome/pull/8009)
> 
> external_components:
>   - source: github://pr#8009
>     components: [ i2c ]

but if i change it to 

```
external_components:
  - source: github://pr#8009
    components: [ dlms_meter ]
```

it works for me! Do i misunderstand something?

pos-ei-don, 2025-02-20:
Fixed in https://github.com/esphome/esphome/pull/8009

SimonFischer04, 2025-02-21:
> [ximex](https://github.com/ximex): I am a bit confused,
> This does not work for me, so i tested the code from
> https://github.com/DomiStyle/esphome-dlms-meter/issues/47#issuecomment-2671025166
>
>
> but [...]

You change is correct. I think just a typo from @ximex .

SimonFischer04, 2025-02-21:
> @mariopenzendorfer
>
> Which provider do you have?
>
>
>
> Yes. Heavily inspired by the dsmr component. Hope this makes implementing more sensors easier / less copy paste.
>
>
>
> Which repo do you mean? Domis [...]

Hi @firegore,

Great so see you here. Big thanks also to you for initially figuring out the evn stuff.

ximex, 2025-02-21:
yes it was a copy/paste error. i copied to codeblock from another PR and forgot to change to component