DomiStyle/esphome-dlms-meter #48: CVE-2025-57808

valki2, 2025-09-02:
In case anyone is still running this code - consider upgrading esphome and rewrite the code to be compliant. Dont run the old one as there is a leak in auth.

valki2, 2025-09-02:
e.g. something like this: https://github.com/valki2/esphome-dlms-meter-2025

SimonFischer04, 2025-09-07:
Hi @valki2 

FYI, I also made a custom component based on the code here a while ago:
https://github.com/SimonFischer04/esphome/issues/1

Which is already successfully used by multiple people.
And mainline PR waiting for review from esphome core team.

Any specific improvements you made?
Would be nice if we could collaborate and get all improvements into one component.